IEEERobotics/bot #244: Senior Design proposal

jasteve4, 2015-01-07:
Can everyone take a look at the senior design proposal.

Any feedback wold be a lot of help.
It will need to be turned in late tomorrow for this semester senior design.

This can be a big win for us.

Thanks.

jasteve4, 2015-01-07:
Here is the link to the google drive:
https://drive.google.com/a/ncsu.edu/#folders/0B_fnIkF7_1pgSkNWTUlEOXg3M28

dfarrell07, 2015-01-07:
I'd add a requirement that it's an Open Hardware and Open Source Software project with solid docs that should allow the community to replicate the board.

jakunesh, 2015-01-07:
I agree with Daniel. Otherwise I fixed a few typos and I think it looks great!

dfarrell07, 2015-01-07:
Maybe some protection-related stuff (/me defers details to ECE folks). For example, it should resist catching on fire if we drop a nut on it.

dfarrell07, 2015-01-07:
PS - I love the idea of proposing this as a Senior Design project. Huge props to whoever came up with that.

PaladinEng, 2015-01-07:
I accept beer in lieu of huge props.

PaladinEng, 2015-01-07:
Do you all think it is feasible/worthwhile to include short/surge
protection so that the BBB is protected?

R00ney, 2015-01-08:
Cool Idea.
Neal

AhmedSamara, 2015-01-08:
Minor terminology dispute, but most of it keeps referring to the connectors (usb, ethernet, hdmi, etc) ports as "peripherals", which I've mostly seen used in the context of add-on devices (ie: the motor cape is a peripheral), so I think that might get confusing, so I'm slightly rephrasing a lot of it to use that.

Also, Spec#7 

> The peripherals shall be on top of the board (ie connectors, buttons, and any interfaces)

Doesn't make sense given spec5

>The cape shall be stackable (i.e. plug in multiple capes into the same beaglebone without issue)   

Since if they're on top of the word, so I think we should remove that one entirely. The only important thing is that connectors and accesses are accessible from the bone, which I think is a given if it's stackable.

jasteve4, 2015-01-19:
Does any know were the flyer image is.

SeanKetring, 2015-01-19:
If you mean the image that I made for the crankarm flyers here is a copy of
that file

AhmedSamara, 2015-01-28:
Nobody wanted us.

closed 2015-01-28 by AhmedSamara