IEEERobotics/bot #245: Print Rail_Adjust_Left

whiteskulleton, 2015-01-08:
Need the Rail_Adjust_Left Printed. Please monitor it as it is a small part and it might jam the printer.
https://drive.google.com/a/ncsu.edu/?tab=wo#folders/0B93g4TjHMS2pZ0NubEFBTWxJZTA

[image]

jakunesh, 2015-01-09:
Today James and Alua printed out 2 beautiful copies of this. Placed on the
top shelf

whiteskulleton, 2015-02-20:
Can I get another one of these printed up?

jakunesh, 2015-02-20:
I just tried printing this in Troxler and I got some strange error in code, I'm going to see if the senior design TA's know what is going on

jakunesh, 2015-03-06:
Printed out 2, they came out very well.

closed 2015-03-06 by jakunesh

whiteskulleton, 2015-03-19:
1. The set screw holes come out slightly too big. It is recommended that they be modeled smaller and then drilled to the correct size for tapping.
2. The part does not made the shaft rigid and allows it to flex some and it is recommended that the part be made more rigid.