IEEERobotics/bot #248: Locating physical pins on the bone that corresponds to a pwm in code.

kvijay1995, 2015-01-11:
I'm having trouble locating the physical pin on the bone that corresponds to a specific pwm in the /sys/class/pwm folder in the bone. For example, there is pwm2, pwm4, pwm5 in the /sys/class/pwm folder, but I'm unable to find the corresponding physical pin for each of these on the bone.

@dfarrell07 , @AhmedSamara , do either of you have an idea of how I could find this?

jakunesh, 2015-01-12:
This is a pin layout of last year that could be helpful. It was moved into the 2014 folder, probably on accident.

https://docs.google.com/spreadsheet/ccc?key=0ArstfDp21akOdE5iQjQ4c1pRU0hQdEYtaUtnZGFJWXc&usp=sharing

closed 2015-05-30 by jasteve4

kvijay1995, 2015-07-31:
So I've found the details needed to solve this problem in the following file:

1) navigate to this folder:
<b> [bot repo on local machine]/bot/setup/fs-common/root/startup </b>

2) Open the <b>pwm.sh</b> file

3) This file assigns the various pins to specific pwm nos. such as:
<b>pwm_to_pin[0]=P9_31</b>

which assigns P9_31 to pwm0 and so on

Other details regarding how this script is run during startup, gpio exports is explained in detail by Daniel in #291