IEEERobotics/bot #249: Print Rail Mount

whiteskulleton, 2015-01-14:
Please print the rail mount. Note - the printing face is the one with all the chamfers.
https://drive.google.com/file/d/0B8cZC9JtfQr6TUNBSkpCYTM1cGM/view?usp=sharing

jasteve4, 2015-01-14:
Done printing, Is there anything else that needs to be printed.

closed 2015-01-16 by jasteve4

whiteskulleton, 2015-03-19:
The rail mount was printed on the correct face.

1. Interesting problem was that the bearing for the ACME rod did not fit in the hole properly and had to be drilled out slightly I believe.
2. The holes for the part that secures the rail was too big. It is recommended that the hole be made smaller and drilled out.
3. The mounting holes that mount the part to the chassis were not sized correctly as the bolts bottom out on the beam.