IEEERobotics/bot #250: Print Grabber Left

whiteskulleton, 2015-01-14:
Please print the Grabber Left on the chamfered side.
https://drive.google.com/file/d/0B8cZC9JtfQr6OTBKY29ZRmhEMTA/view?usp=sharing

jasteve4, 2015-01-15:
done

closed 2015-01-16 by jasteve4

whiteskulleton, 2015-03-19:
This part was printed incorrectly and was printed from the bottom face.
However, it is interesting to note that the hole for the sleeve bearing for the ACME rod was a perfect fit whereas the exact same hole on the rail mount didn't fit correctly even thought that part was printed correctly.

Problems:
1. The mounting holes to mount this part to the chassis were not the correct depth and allowed the bolts to bottom out.
2. The white sleeve bearing hole had too much interference and had to be sanded down. This is also interesting because the Gripper_1 had the exact same hole but was printed from a different face and it fit perfectly.
3. It is recommended that the front bottom edge be filleted so that it will not snag on anything if it hits the ground.