IEEERobotics/bot #251: Print Grabber_1

whiteskulleton, 2015-01-14:
https://drive.google.com/file/d/0B8cZC9JtfQr6YVlieHJFUWtkY3M/view?usp=sharing
[image]

jasteve4, 2015-01-15:
done

closed 2015-01-16 by jasteve4

whiteskulleton, 2015-03-19:
The grabber was printed from the wrong face. However, it did turn out quite nicely for as the linear bearing fit perfectly into the hole and did not need any set screws.
The white sleeve bearing was also a perfect fit and did not need a set screw.

Problems:
The arm has some wobble in about the linear bearing due to slop between the threads of the nut and the shaft and due to the fact that the nut is not a transition fit in the plastic. A hole was drilled at an angle and a set screw was added to remove the slop between the plastic and the nut but there is still some slop. This slop is a sort of hinderance as it is hard for the arm to pass under the brackets that hold the open beam.