IEEERobotics/bot #252: Print Stepper Block

whiteskulleton, 2015-01-14:
I will need 2 of these stepper blocks printed. Print one for now then print the other objects and come back and print the second one.
https://drive.google.com/file/d/0B8cZC9JtfQr6c2NmeUJRbHNFRk0/view?usp=sharing
[image]

jasteve4, 2015-01-15:
done

whiteskulleton, 2015-01-22:
I need another one of these printed

cdyoder, 2015-01-23:
Currently being printed.

cdyoder, 2015-01-23:
Printed. Decent quality (definitely usable). Placed in the cabinet in "Mark's Box".

closed 2015-01-23 by cdyoder

whiteskulleton, 2015-03-19:
Problems:
All the holes came out as ovals. This is bad because when some of the holes are tapped, there will be little thread engagement and they won't hold a screw properly. It is recommended that the holes be modeled smaller and then drilled out to get to the correct size.

The printed model came out to the correct size.