IEEERobotics/bot #260: Testing the DC Motor for turning the Rubix Cube

cdyoder, 2015-01-21:
Currently, we have a DC motor with encoder mounted on the bot to turn the Rubix cube  top row. There is discussion of switching the motor to a stepper motor instead of the DC motor. Before we make the switch, I would like someone to write a basic code, using the Bone, to send a signal to the DC motor and see if we can reliably rotate the Rubix cube. 

Ideally, I would like this to be done from a Bone using the DC motor currently mounted to the bot. If you're interested, please talk with either myself or Ahmed.

cdyoder, 2015-01-21:
@pvsteja 
@asatish127

AhmedSamara, 2015-01-21:
The EE overview of what you'll need to do is:

-First see how accurately and consistently the motor can be moved using a power supply and just turning it on/off by hand.
-If that works well enough that it seems like it's even worth attempting, you can move to using GPIO pins (either on a Beaglebone or on an Arduino) to control a transistor that controls the supply of power to the motor.

cdyoder, 2015-01-21:
Update:
The DC motor is able to turn the top row of the cube. Proven 1/20/15.

jakunesh, 2015-02-11:
according to above comment:
"The DC motor is able to turn the top row of the cube. Proven 1/20/15."
this issue is closed. Using a micro controller to do this will be a separate issue, since that is basically solving the task

closed 2015-03-18 by jakunesh