IEEERobotics/bot #261: Reminder for all programmers: Set your text editors to replace tabs with 4 spaces.

AhmedSamara, 2015-01-21:
As a general rule of thumb for programming, tabs are undesirable because their actual size will vary depending on the resolution and text editor settings of the person viewing, and it can make it particularly annoying to look at.

In python Tabbing is especially critical because "levels of indentation" are the python equivilant to brackets in C, and thing having 1-tab+1-space, or 2 spaces, or 2 tabs would be the same level despite not looking at aligned at all.

Long story short:
The convention we've agreed on for this project is to use 4 spaces per level of indent.

Here are some guides for common text editors:

Gedit:
`edit -> preferences`
Check "insert spaces instead of tabs".

Notepad++:
`preferences -> Language/menu tab`

Vim:
edit `~/.vimrc` and insert:
`:set expandtab
:set tabstop=4`

dfarrell07, 2015-01-21:
+1 to all of that, super important for all Python programming.

> The convention we've agreed on for this project is to use 4 spaces per level of indent.

Not just us, the entire Python world via [PEP8](http://legacy.python.org/dev/peps/pep-0008). There's even a section called "[Tabs or Spaces?](http://legacy.python.org/dev/peps/pep-0008/#tabs-or-spaces)".

> Tabs or Spaces?

> Spaces are the preferred indentation method.

> Tabs should be used solely to remain consistent with code that is already indented with tabs.

> Python 3 disallows mixing the use of tabs and spaces for indentation.

> Python 2 code indented with a mixture of tabs and spaces should be converted to using spaces exclusively.

> When invoking the Python 2 command line interpreter with the -t option, it issues warnings about code that illegally mixes tabs and spaces. When using -tt these warnings become errors. These options are highly recommended!

dfarrell07, 2015-01-21:
> >The convention we've agreed on for this project is to use 4 spaces per level of indent.

> There's even a section called "Tabs or Spaces?".

And another section on [Indentation](http://legacy.python.org/dev/peps/pep-0008/#indentation).

> Use 4 spaces per indentation level.

AhmedSamara, 2015-01-30:
Hopefully you've all done this by now.

closed 2015-01-30 by AhmedSamara