IEEERobotics/bot #264: Block Level Diagram and Wiring Info for the Bot

SeanKetring, 2015-01-21:
We need a fully detailed wiring scheme for the bot's power system, and a block level diagram showing how everything fits together.  This issue can be closed after this info has been uploaded the wiki.

SeanKetring, 2015-01-23:
[image]

Here is a crude illustration of the power system, more info to come later but so far there is a basic color scheme:

Black is going to always be ground, the color with black will be the higher line

Battery voltage feeding into D.P. will be blue/black and red/black for logic battery and motor battery

All motors used in the task actuators will be powered from D.P. using red/black 22 gauge wire

All servos powered by D.P. will be using a yellow/black color scheme

The beagle bone black will be powered using blue/black wire 

The ADC will be powered with Green/Black wire.

Still need a diagram illustrating the interface with D.P.

SeanKetring, 2015-01-28:
Here a map for interfacing with Dark Power. 

[image]

SeanKetring, 2015-01-28:
Note, Dark Power has a direct connection to the capes, these wires are blue and black

SeanKetring, 2015-01-30:
In the interest of keeping track of changes, beagle bone black is powered through a cable that's all black, and two wires, kind of like a wall wart.  

The servo may have to be powered by the logic battery, unless we can figure out a way to share one d.p. Output between two devices 

Still to do as of when I left today:

Split the battery connections from two two pairs of wire to four two pairs of wire

Figure out which order connect to logic wires on the drive motors, to the capes

Mount and wire the adc 
Make logic diagram, 
Wire logic

SeanKetring, 2015-02-10:
[image]

A slightly more accurate block diagram

SeanKetring, 2015-02-10:
[image]
 
Revised Dp guide

SeanKetring, 2015-02-10:
[image]

Simon says sensor board

[image]

Stepper motor control board

SeanKetring, 2015-02-10:
[image]

Updated version

SeanKetring, 2015-04-03:
[image]

jasteve4, 2015-05-30:
This has all been moved to a wiki

closed 2015-05-30 by jasteve4