IEEERobotics/bot #270: Course Creation Procedure

jdekler, 2015-01-24:
Currently we have a small jar of pinkish white paint and a larger can of Black primer. We have all the components for the course, however the border walls will have to be taken down. 

Procedure:
1. Remove LED lighting
2. Coat entire surface with black primer
3. Mark out start box and play boxes with tape and fill with white paint
4. lay out course in painter's tape specified by rules and paint white
5. Lay down finish line
6. Assemble

SeanKetring, 2015-01-25:
will there be a time that you will be working on this?  I have access to
some power tools and painting supplies

psychomugs, 2015-02-08:
Removed the LEDs and painted the black base coat, currently waiting for it to dry.

Josh and Ahmed brought out a large, one-piece platform that is apparently better for testing line following. Painting that can take place another day when we get more black paint.

cdyoder, 2015-03-13:
The final course (the course with walls) has been finished and approved for testing. The other course (two sheets of plywood) can also be used for testing if needed.

jakunesh, 2015-03-18:
I believe @cdyoder last update qualifies as this issue being closed until proven otherwise

closed 2015-03-18 by jakunesh