IEEERobotics/bot #271: FOSS fair 2015!

AhmedSamara, 2015-01-26:
Hey guys, if you're interested, FOSS (Free Open Source Software) fair will be held Feb 7 , in EBII.

If you're interested, here's the website:

http://lug.ncsu.edu/fossfair/FossFair2015

In particular, I think the "What it means to be a maintainer" and both of @dfarrell07's talks are worth attending.

dfarrell07, 2015-01-27:
> In particular, I think the "What it means to be a maintainer" and both of @dfarrell07's talks are worth attending.

+1, full of agree and not just because they're my talks. Clearly relevant to the team's CS work, important topics generally.

https://github.com/IEEERobotics/bot2014/blob/master/Vagrantfile
https://github.com/IEEERobotics/bot2014/blob/master/Dockerfile

AhmedSamara, 2015-02-08:
Foss fair was awesome and everyone that didn't go is dead to me.

closed 2015-02-08 by AhmedSamara