IEEERobotics/bot #278: 2015 pilot dev state machine.

AhmedSamara, 2015-02-06:
I've started scribbling down an FSM similair to [last years](https://docs.google.com/a/ncsu.edu/drawings/d/1yCgWLLP2UAkc1H-t0DCxxHkTV1321iQYH9oK4q20sns/edit?usp=drive_web), but I think this one might be simple enough to just do linearly.

ie:

line-follow until T-intersection. stop.
turn.
line-follow to block. 
(solve activity)
turn around.
line-follow to intersection.

(repeat to finish)

AhmedSamara, 2015-03-04:
http://www.ewh.ieee.org/reg/3/southeastcon2015/track/index.html

This complicates things a lot. As it turns out, they will have intersections, so the state machine will have to account for this.

AhmedSamara, 2015-03-08:
Some functionality that I would like is:

- "follow_to_intersection": 
- Ability to differentiate intersection from right angles. (We'll avoid worrying about 4-ways for now, because I'm positive that the demo boards having them was a mistake).
 - Ideally, this would be handled within the follow function so that the previous function can continue working uninterrupted.

- "follow_to_box"
 - Functionality would be similar to other follow_to_intersection. I think it might even be possible to just use follow_to_intersection, since to the IR's might not even see them as different.
It's worth noting that this function doesn't even have to tell the difference between a box and an intersection, because neither one would ever be called in a situation where it might hit the other.

- turn_on_intersection 
 - This should be easy enough to do without any huge changes. I'm working on a function in mec_driver for rotating 90 degrees, but some line-detecting intelligence in it would be useful if that's not accurate enough.

AhmedSamara, 2015-03-08:
Update: turn_on_intersection is unecesary, blindly turning works fine and it just needs to get close enough for IR to see it anyway.

AhmedSamara, 2015-03-09:
All that matters is being able to follow the line (including around corners) and stop whenever you hit an intersection or box.

Return values aren't important, but if you return information strings about what exactly it saw to stop, that has the potential to be useful.

jasteve4, 2015-05-31:
Pilot will need to be updated for general navigation.

closed 2015-05-31 by jasteve4