IEEERobotics/bot #280: Finish assigning BeagleBone black pins.

AhmedSamara, 2015-02-08:
Still to be assigned:

Signal | type | Suggested pin | notes
Rubix cube gripper motor:  | GPIOs |  | presumably connected to an h-bridge on PCB, or motor cape
Simon says servo: | PWM | P9_21 (EHRPWM0B) |
Etch-a-sketch stepper (left) | PWM | P9_14 (EHRPWM1A) | stepper motors previously required 4 GPIOS, but the new design by Josh lets us use one PWM.
Etch-a-sketch stepper (right) | PWM | P9_16 (EHRPWM1B) | (directly next to the other etch-sketch pwm. Easier wiring

All of these assignments were pretty much picked arbitrarily, but unless anyone (Particularly @jasteve4 ) has any assignments or designs/code with other assumptions or a reason to use ECAP PWM, over EHRPWM I'm going to go ahead and use these.

AhmedSamara, 2015-02-08:
This has to change, the rubix gripper actually requires two GPIOs for controlling direction, and doesn't really need a PWM.

jasteve4, 2015-05-30:
This issue can be closed, this years bot will not need gripper.

closed 2015-05-30 by jasteve4