IEEERobotics/bot #287: Sorting off individual tasks as Undergraduate research projects.

AhmedSamara, 2015-02-19:
http://undergradresearch.dasa.ncsu.edu/funding/nc-state-summer-undergraduate-research-grants

Applications for summer undergraduate research grants is open, and I think this would be a good way to get funding as well as a good way for members to "double-dip" their time since whatever you work on could be a research project as well as contributing to the team.

Let me know if you have any ideas.

AhmedSamara, 2015-02-19:
Of-course, this is just for undergraduates. Grad students can still be a part of the project, but you can't be on the application and it still has to mainly belong to one of the undergrads.

PaladinEng, 2015-02-19:
Great idea. Let's get a list of projects together as well as interested
contributors to tackle this. I am willing to work/support all the projects
however I can. The deadline is coming up fast, so we need to sort this out
quickly.

SeanKetring, 2015-02-19:
I can't say for sure if I'll be in Raleigh for the summer, but if I am I am
totally interested in doing one of these projects.   If I'm in Charlotte
for an internship then I'd be interested in doing something remotely.

jakunesh, 2015-02-20:
The stepper motor controller sounds like an excellent candidate for this, and I think would be very reasonable for a summer project

jakunesh, 2015-02-20:
I think a PCB of the flyback converter project is also a great idea

dreamphantom, 2015-02-20:
Unsubscribe

whiteskulleton, 2015-02-20:
I thought of doing something like micro-pneumatics or micro-hydraulics for really small parts or for around electronics.

AhmedSamara, 2015-02-20:
I think the stepper motor controller is too small to be a real project.

The design is already fleshed out and complete, and all that's left is to
make a PCB if that's needed and possibly make it as a breakout-board.

jakunesh, 2015-02-20:
I'd assume all EE projects would end in a PCB, and although it isn't a advanced circuit for a senior, it'd be an awesome learning experience for a freshman or sophomore, remember we didn't use 555 timers to blink LEDs till ECE212. I suppose it depends on what they mean by research.

Check out this link https://www.tindie.com/products/TEMProducts/stepperstack-multi-purpose-stepper-controller/

Wouldn't that be a bad-ass thing to implement in a cape for the bone? It would be a full on open-source hardware project when finished too

AhmedSamara, 2015-02-20:
I liked the creation of a Beaglebone MSP430 cape that Josh suggested.

It would essentially be a simplified version of the senior design project
that we proposed before.

It would involve designing a daughter-board for the beaglebone, which takes
commands from the beaglebone via i2c, and exposes all of the IO of the
msp430 to another set of headers.

This would also be a good excuse to get funding for another beaglebone.

jakunesh, 2015-03-01:
A reminder that the deadline for this is coming up. I know there are a few us us with considerable interest. If there is a project you want to do, the time to move and create a proposal is now, remember there are both individual and team proposals.

PaladinEng, 2015-03-02:
The projects I'm tracking are:

1) The stepper motor controller cape
2) The flyback converter cape
3) A network cape to replace our current wifi-setup

Does anyone specifically want to tackle one of these projects? We need to
be able to attach names to the applications.

SeanKetring, 2015-03-02:
I'm interested in taking on proposal 2

jakunesh, 2015-03-27:
Missed the boat fellers, better luck next semester.

AhmedSamara, 2015-05-21:
Hey everyone, grants for the fall close June 22.

http://undergradresearch.dasa.ncsu.edu/funding/nc-state-academic-year-fall-and-spring-undergraduate-research-grants

If you have any ideas, please make a document in the 'project proposals' folder in the google drive.

SeanKetring, 2015-06-23:
Research grants have now closed for the summer

AhmedSamara, 2015-06-24:
They just extended the deadline until the 29th if anyone has any last minute ideas.