IEEERobotics/bot #291: GPIO pin export on the BeagleBone

kvijay1995, 2015-03-04:
I'm having trouble finding the script that automatically exports the pins for GPIO usage for us when the bone boots. 

For e.g. I wanted to export pin 31 as gpio so I did this:

echo 31 > /sys/class/gpio/export

But when I rebooted the BeagleBone, this specific pin is not exported anymore. I want to be able to do this automatically for a variety of pins at startup, rather than manually enter the command each time.

@AhmedSamara , @dfarrell07  Do you guys have any idea where I can find this?

dfarrell07, 2015-03-04:
> I want to be able to do this automatically for a variety of pins at startup, rather than manually enter the command each time.

[This script](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/root/startup/setup_all.sh) is designed to provide that behavior. You will likely be able to follow/copy the examples that already exist in that file (they are well documented), but I can give more details about how that all works if needed. :)

dfarrell07, 2015-03-05:
> details about how it all works if needed. :)

I'll just give the details anyway.

[This chunk of `setup_bone.sh`](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/setup_bone.sh#L56-L76) copies various configuration files (version controlled in our repo) to the correct locations on a bone, and optionally does the same for an additional bot-specific set of config files.

```bash
echo "========================"
echo "Populating filesystem..."
echo "========================"
chown -R root.root fs-*
rsync -va fs-common/* /
chmod 755 /root/.ssh
chmod 600 /root/.ssh/id_rsa
chmod 644 /root/.ssh/authorized_keys
# We need to customize uEnv.txt for each install
ROOT_UUID=$(blkid -t LABEL=rootfs -o value -s UUID)
sed 's/{{ROOT_UUID}}/'$ROOT_UUID'/' /boot/uboot/uEnv_template.txt > /boot/uboot/uEnv.txt

if [ "$bot_specific" = true ]; then
    rsync -va fs-bot/* /
    echo "============================"
    echo "Enabling bot wireless uplink"
    echo "============================"
    echo Enabling uplink...
    ifup wlan1
fi
```

Note that [this line](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/setup_bone.sh#L61) overlays the files in our [`bot/setup/fs-common`](https://github.com/IEEERobotics/bot2014/tree/master/bot/setup/fs-common) repo directory onto the bone's file system.

```bash
rsync -va fs-common/* /
```

Which puts out [`bot/setup/fs-common/etc/init.d/bbb`](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/etc/init.d/bbb) script in a location that Linux will, by default, access at boot. The `bbb` init script contains [this code](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/etc/init.d/bbb#L14):

```bash
case "$1" in
start|reload|force-reload|restart)
        /root/startup/setup_all.sh
        ;;
stop)
        exit 0
        ;;
*)
        echo "Usage: $0 {start|stop}"
        exit 1
        ;;
esac
```

Which in turn calls the [`setup_all.sh`](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/root/startup/setup_all.sh) script I [mentioned above](https://github.com/IEEERobotics/bot2014/issues/291#issuecomment-77277081).

It [sources](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/root/startup/setup_all.sh#L3-L7) a bunch of [helper functions in its dir](https://github.com/IEEERobotics/bot2014/tree/master/bot/setup/fs-common/root/startup):

```bash
dir=$(dirname $0)
source $dir/slots.sh
source $dir/pwm.sh
source $dir/gpio.sh
source $dir/adc.sh
```

Which provide it nice helpers like `load_gpio`:

```bash
# Gun laser: P8_35 (gpio8)
load_gpio 8

# Gun motor - left: P8_37 (gpio78)
load_gpio 78
```

This [helper](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/root/startup/gpio.sh#L3-L14) does the same export process [you described](https://github.com/IEEERobotics/bot2014/issues/291#issue-59882001):

```bash
gpio_sysfs=/sys/class/gpio

load_gpio () {
  num=$1
  if [ ! -e $gpio_sysfs/gpio${num} ]; then
    echo $num > ${gpio_sysfs}/export
  else
    echo "GPIO $num already exported"
  fi
  echo out > ${gpio_sysfs}/gpio${num}/direction
  echo 0 > ${gpio_sysfs}/gpio${num}/value
}
```

So you can add setup calls in [`setup_all.sh`](https://github.com/IEEERobotics/bot2014/blob/master/bot/setup/fs-common/root/startup/setup_all.sh):

```bash
# My new thing foo needs GPIO 77 because bar
load_gpio 77
```

And Linux will [call them for you at boot](http://stackoverflow.com/a/12973826/1011749). :)

kvijay1995, 2015-03-05:
Haha! oh gosh, thank you for that comprehensive explanation. I was curious about how the script got called by the OS at boot, so I do appreciate you going out of your way to explain this to me.

That more than answered my question, thanks Daniel! :)

closed 2015-03-05 by kvijay1995

dfarrell07, 2015-03-05:
> so I do appreciate you going out of your way to explain this to me.

Glad to help. It's good to have it documented. :+1: