IEEERobotics/bot #293: I2c object for the i2c addressable pot

jasteve4, 2015-03-06:
There needs to be an abstraction class created for the i2c addressable pot.

AhmedSamara, 2015-03-07:
Using the i2cObject library is probably overkill, smbus should be more than enough.

What's the part number for the pot?

jasteve4, 2015-03-09:
Have trouble with passing hex values form the cli client to the function set_pot_wiper.

jasteve4, 2015-03-09:
Fixed it, must pass variables in as int. Hex values are treated as a string.

jasteve4, 2015-03-10:
The Pot object is done and read for use. The find_ambient_light function can be used to set the pot to the current ambient light. More useful information can be found on the on the senors cape wiki.
 https://github.com/IEEERobotics/bot2014/wiki/Sensor-Cape

closed 2015-03-10 by jasteve4