IEEERobotics/bot #294: Meeting Over Spring Break

AhmedSamara, 2015-03-08:
Traditionally the club gets a lot of dev work done over spring break. With the competition quickly approaching, and with no class it is a great opportunity to get a lot of work done.

Those who are willing and able to come in and work are highly encouraged, and I expect there will be dev work done every day of break. This issue can be used to coordinate time between members so that we can make sure no one has to work alone.

Starting with tomorrow @PaladinEng has promised to provide some donuts and coffee in the morning (~10am).

Feel free to post when you plan to come in and give an approximate time. Personally myself and @jakunesh plan to be here every day.

PaladinEng, 2015-03-08:
Donuts and coffee have arrived. All we need now are some teammates.

jakunesh, 2015-03-10:
I plan on coming in around 11 tomorrow, Tuesday (3/10)

closed 2015-03-17 by AhmedSamara