IEEERobotics/bot #299: Linear actuator solenoids for simon says player.

AhmedSamara, 2015-03-11:
Yesterday Shep recommended pairs of linear actuator solenoid with 1'' throw that were about $14. 

Because these are incredibly easy to work with, compared to the servo arm solution that @kvijay1995 has been working on and running into issues because of, I think these would be a great buy that we should bite the bullet on ASAP (and even if they don't end up being used this year they'll be useful again in the future).

The only problem is that I can't find them, does anyone have more information on what he was talking about? 
@jasteve4 @cdyoder @PaladinEng 

Or have his contact info so we can ask him?

PaladinEng, 2015-03-11:
Sorry, I don;t know which ones he had in mind, and I don;t have his contact
info

cdyoder, 2015-03-11:
I have a few leads on solenoids with a 1 inch throw but not for ~$14. I'm assuming we prefer 12VDC over 24 or 36VDC?

McMaster:
http://www.mcmaster.com/#solenoids/=w9m0ol

Sparkfun:
https://www.sparkfun.com/products/10391

Digikey:
http://www.digikey.com/product-detail/en/F0452A/527-1017-ND/668302

jasteve4, 2015-03-11:
Here is a link to a website he gave me. 
http://www.jameco.com/Jameco/catalogs/c112/P104.pdf

I also found this website, but these parts are a bit expensive. most of these parts can be found at newark. we can get these pars by the beginning of next week at stander shipping. 
http://electromechanicalcomponents.guardian-electric.com/category/dc-tubular-solenoids/

how long does the stroke need to be? And I can try to narrow it down. 1" stroke is very hard to find can 1/2" work.

jakunesh, 2015-03-27:
We are totally not doing this. Although we do seem to have some mystery solenoids now...

closed 2015-03-27 by jakunesh