IEEERobotics/bot #303: BeagleBone not turning on when Stepper motors plugged in.

AhmedSamara, 2015-03-13:
A strange bug.

The BeagleBone on the competition bot doesn't turn on when the stepper motors are plugged in.

It works when the stepper motor is unplugged, and it stays on when it's plugged in after if it's already on.

We've already tried:
 - Initializing the GPIO pins controlling the stepper motor to outputs explicitly.
 - Running the startup script.
 - Plugging in the motor battery in addition to the logic battery.

PaladinEng, 2015-03-13:
Did you check the current draw with and without the steppers attached? I
suspect a brownout. BBB is very sensitive to power demands on startup

AhmedSamara, 2015-03-13:
That sounds like exactly what this is. How do we check the current draw?

PaladinEng, 2015-03-13:
I would start by measuring the current through the voltage supply line
during startup with and without steppers. Alternately, if you have a some
sort of "enable" switch set up so that you can make sure everything stays
off during startup, then you can wait until after a successful boot to
power everything up

AhmedSamara, 2015-03-18:
A brown-out still sounds like the most likely setup.

The original setup script set all GPIOs as outputs with a value of 0, and changing them all of them to inputs (high-impedance) to prevent any accidental current draw, but that also did not solve the issue.

We found this [article](http://elinux.org/Beagleboard:BeagleBoneBlack) that sounded relevant:

>The onboard HDMI driver uses the LCD pins. They are still connected to the expansion header, but in some cases, they could cause an LCD cape to have some noise issues unless the cape has a buffer.
A buffer should take care of this issue. If you want to use these pins as GPIO pins, you may be limited in speed due to the capacitance caused by the loadig of the pins by the HDMI driver.

I don't know exactly how this relates to us, but the GPIOs connected to the LCD are known to cause problems, and we found that the bone could boot up normally once it was plugged into different gpio pins. (specifically, gpio pins related to the left etch-a-sketch gripper).

It should be easy to solve this problem just by making sure all etch-a-sketch motor assignments are not on LCD pins.

It would be a good idea to make note of this in the future, because even though for most GPIO uses speed is not a huge issue, it's absolutely critical for stepper-motors.

@kvijay1995 can you go ahead and assign new pins to this when you get the chance? I can do it but I don't want to yank the rug out from under you if you're working on something.

PaladinEng, 2015-03-18:
Is the HDMI cape overlay disabled?

AhmedSamara, 2015-03-18:
Yes. But according to the wiki, just their capacitance can still cause
problems for them in gpio mode.

jakunesh, 2015-03-27:
Is this confirmed solved?

jakunesh, 2015-04-06:
Confirmed solved. We now have both steppers motors running, controlled by the bone.

closed 2015-04-06 by jakunesh

AhmedSamara, 2015-04-28:
Just a note: We didn't actually solve this problem, we just avoided it by using pins that weren't attached to the HDMI.