IEEERobotics/bot #305: After a clean install of Debian, not able to interface with the niks

jasteve4, 2015-03-14:
After flashing the newest Debian image, I am have some difficult interface with the wireless niks.

Here is what I have done up to this point.

I wget the newest version of Debian image: Here is the link to the beaglebone Debian wiki where I got the image. BBB Rev C (4GB eMMC) Flasher: (lxde-4gb)
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

Here is the version that was flashed on the the bone: BeagleBoard.org Debian Image 2015-03-01

Next, I run the apt-get update && apt-get upgrade -y
This runs fine as far as I can tell.

next, I follow the BBB setup wiki, https://github.com/IEEERobotics/bot2014/wiki/BBB-Setup, no problem here.

Next, I follow the Dev wiki, https://github.com/IEEERobotics/bot2014/wiki/Dev-Environment-Setup#clone-IEEERoboticsbot2014-and-pybbb-submodule.

Next, I ran the wireless setup script and rebooted the bone and connected to the internet via wlan1. I can still ping google.com.

jasteve4, 2015-03-14:
I Checked my mac address on the nomad and the the mac address is not registered, I think this can be the sources of most of my problems. 

I don't know how to add my mac address without spoofing the bone's mac address. I remember talk about a way to add the mac address via manual. This would be good place to document, or add it to the bone setup wiki or some other place.

jasteve4, 2015-03-14:
Add mac address to the nomad fixed the problem with nics, I can now do apt-get update, but the nics do not come up after restart or power up. I remember the nics should come up before running the -b flag  in the setup_bone.sh script. Not sure what the problem could be coming from.   

After reset, I have to run ifup wlanx to get the nic comes up. Also mon.waln0 doesn't show up when ifup wlan0 is ran.

jasteve4, 2015-03-17:
I Think there is something wired going on with Debian 7.8. I was able to get wireless working with Debian 7.4, the factor build, but there was problem with apt-get update && apt-get upgrade -y.

Does anyone have Debian 7.7 or 7.6 image that can be used to flash an sdcard.

jasteve4, 2015-03-17:
I have replaced the bone on the bot with a bone that has a Debian 7.7 image and the wifi is back up and running. I used the bone that VJ was using. This issue will remain open until we figure out how to get the nics working on Debian 7.8.

jasteve4, 2015-05-31:
Due to the number of bones that have been destroyed, this problem needs to be solved.

I think that a service needs to be written to bring the nics up on boot up, but I have no experience with this.

jonathonwpowell, 2015-06-30:
Link to register a device with nomad manually:
https://oit.ncsu.edu/resnet/register

closed 2015-09-30 by jasteve4