IEEERobotics/bot #306: line following algorithm

jasteve4, 2015-03-14:
I have been able to get the robot to follow a straight line with minimal error. The real problem is detecting an intersection and turns.

I have been trying to read the side sensors to know when an intersection or turn is present. The side sensors are producing false positives. So, I am getting a intersection and turns confused and what side the intersection/turn is on.

I have been doing a lot of reading online and most line following robots only use a single ir array to follow an line. Using one ir array may make things easier. I found a document, https://www.pololu.com/file/0J195/line-maze-algorithm.pdf, that may be helpful for both pilot and Follower. 

This issue will close once intersection and turns can be detected without error.

AhmedSamara, 2015-03-27:
as of: 55cd4fb3f904fd774ce5826610c587cdc5217b9d

Follower can now follow and stop at turns/intersections. More importantly, it accurately stops with the intersection being centered on the (square) array!

Which means that detecting the difference between a turn/intersection is now trivial, and pilot can and should be able to safely make its way to different tasks.

I think it's safe to say that this issue is closed!

closed 2015-03-27 by AhmedSamara

dfarrell07, 2015-03-27:
Huzzah! Great job all! :D

AhmedSamara, 2015-03-28:
I just realized how appropriate it is that this issue was #306.

ECE 306 is the Embedded Systems class where we build a line-following robot.