IEEERobotics/bot #308: Contact info request

cdyoder, 2015-03-18:
Today, Mark and I spoke with a representative from Autodesk regarding the use of Fusion 360 and myhub in the design process (for those unfamiliar with myhub, think of myhub as the mechanical equivalent of github). In our conversation, the representative expressed interest in coming to NC State to present on myhub and good CAD design practice. He has asked for the contact info of organizations of students here on campus who may or may not be interested in hearing his presentation (ex: IEEE, ASME, etc). 

If you know of anyone in a leadership position, or in a faculty mentor position, who may be interested, please list them below in a comment so we may seek them out for comment. Also, if anyone is familiar with the required process for bringing a company to campus for a lecture, or knows the point of contact within the university for such matters, please post that name below as well. 

Having the speaker come and present is a big opportunity and we do not want to let it go to waste.

jakunesh, 2015-03-19:
I talked to my professor Kevin Sutton (kgsutton@ncsu.edu) who seemed very interested in helping to get the word out about the talk, he sent me to Nolan Fahrer (nefahrer@ncsu.edu) who now teaches the architectural classes in the college of education using Autocad.  He is also willing to send out emails, make announcement in classes and get me in touch with how to reserve a room through the college of education (which is different than the college of engineering)

Organizations to reach out to might include the Arial robotics club, underwater robotics club, the MAE department and I also think the Entrepreneurship initiative would be interested.

AhmedSamara, 2015-03-19:
Is there a minimum number of students they'd need for it to happen?  
So that we know how what type of classroom to reserve.

closed 2015-05-30 by jasteve4