IEEERobotics/bot #310: No module named bbb.gpio when start server on BBB connected through USB

ChadZhongqiangChen, 2015-03-18:
I pulled the code from Master branch, merge the branch into the etch_a_sketch branch, and used rsync to load the entire directory to the BBB connected through USB. When I tried to start the server on the BBB, I met the following error:

Starting server
Starting CLI
Traceback (most recent call last):
  File "./bot/server/ctrl_server.py", line 15, in <module>
    from bot.follower.follower import Follower
  File "/root/bot2014/bot/follower/follower.py", line 9, in <module>
    import bot.hardware.ir_hub as ir_hub_mod
  File "/root/bot2014/bot/hardware/ir_hub.py", line 6, in <module>
    import bbb.gpio as gpio_mod
ImportError: No module named bbb.gpio

AhmedSamara, 2015-03-18:
Is this on the bone Josh just set up?

If so, just run these commands.

`cd ~/`   
`git clone git@github.com:IEEERobotics/pybbb.git`   
`cd pybbb`   
`python setup.py install`   

This should install the pybbb library. Do the same for any other modules you have issues with.

AhmedSamara, 2015-03-20:
Did this solve your problem @ChadZhongqiangChen ?

closed 2015-04-28 by AhmedSamara