IEEERobotics/bot #311: H-bridge boards for Simon Says player

jasteve4, 2015-03-18:
To support the five motors that are need for the Simon Says player, we need to create a motor controller board. I have two types of H-bridge ics. 

DRV8833
http://www.ti.com/lit/gpn/drv8833 

DRV8848
http://www.ti.com/lit/gpn/drv8848

It doesn't matter which ic we use. Each motor will need 2 GPIOs. This can be simulated by tying a pin to power or ground.

jakunesh, 2015-03-18:
The motors have arrived

AhmedSamara, 2015-04-02:
The basic circuit worked on a breadboard, but not when moved to a proto-board. We're pretty sure the issue is that nsleep was left floating, which is required to enable the H-bridge. 

@jasteve4 wired it correctly.

Here is what we believe is the current (and correct) configuration:

[image]

AhmedSamara, 2015-04-02:
Also, another note: When we were debugging we continuously find a short between Vm and Vref.  
After a few hours of trying to find the issue on the board, we fixed the problem mentioned above and it started working again.  
The short is apparently internal and perfectly normal.  
The only thing that should be needed to correct the rest of them is tying nsleep to 3.3v

jakunesh, 2015-04-04:
note: that diagram is incorrect. nFault is an output and should not be tied to anything except a GPIO if it's being used.

closed 2015-04-28 by AhmedSamara