IEEERobotics/bot #314: Current state of the to buy list ? ?

jasteve4, 2015-03-18:
I could not find an issue about the current state of the "to buy list". So, this seem to be a good place to start this issue. 

I have added some parts form newark and pololu to the current order page on the to by list.

jakunesh, 2015-03-27:
Here's the deal. Competition is right at 2 weeks away. Anuj thinks it will be fastest if we order the parts ourselves and then we can get reimbursed for them. 

Those willing go ahead and get the parts ordered to your homes and you will be reimbursed.

MAKE SURE YOU POST HERE THAT YOU ARE GOING TO BUY SOMETHING BEFORE YOU DO! We don't want any double orders

AhmedSamara, 2015-03-30:
I'm about to send out the order from pololu for micro-motors and IR-arrays.

Is that all we need from them?

AhmedSamara, 2015-03-30:
Order sent.

AhmedSamara, 2015-03-30:
Ordered sdp as well.

jakunesh, 2015-03-30:
I am making the purchase for the Newark parts now

jakunesh, 2015-03-30:
All the parts that were on the To Buy list have now been ordered. This can be considered that last chance we had to order before SEcon, lets hope it actually comes in

closed 2015-03-30 by jakunesh