IEEERobotics/bot #316: MAC address register with ncsu

jasteve4, 2015-03-19:
We need to register the mac address of the nics with the club. I need to make room for some of my devices.

AhmedSamara, 2015-03-26:
I thought the MAC address was specific to the individual Beaglebone?

Regardless, The website for manually registering them is [here](http://webapps.ncsu.edu/cyclone/campus/registration/device/).

And you can find your MAC address manually via `ifconfig`, but since you already have them registered, just go into your [nomad registration](https://nomad.ncsu.edu/) and copy the MAC addresses and I'll register them for you.

dfarrell07, 2015-03-26:
> I thought the MAC address was specific to the individual Beaglebone?

MAC addresses are always specific to L2 NICs.

jasteve4, 2015-05-30:
Thanks Ahmed and Danial, we should move this to a wiki so it will be easy to find when we are setting up the nics on any new bots.

closed 2015-05-30 by jasteve4