IEEERobotics/bot #329: CS bot is on fire. again.

AhmedSamara, 2015-03-27:
After confirming that the motors were all plugged in and working properly, we plugged the IR arrays into the competition bot.

After turning it on, there was a burst of smoke. We couldn't identify the exact source, but it's likely that it was somewhere from the ADC board.

After removing the IR arrays again, we were able to confirm that the motors could still drive normally.

The original source is still unknown, but we're debugging now.

jakunesh, 2015-03-27:
The source of the smoke was the P-fets on the ADC board, @jasteve4 and I were able to confirm this and replace that component.

Thinking it might of been a shorted IR array and the P-fet had done it's job to protect the circuit, since we were still able to read the ADCs, we replaced the P-fets and tried again on the CS bot using a known good IR array.  It promptly smoked.

Having confirmed this was a problem on the board we probed the connectors of the ribbon cables and found that between the first and second iteration of the board the connectors had been flipped.

I then flipped the connectors so that they would be able to be used with the CS bots ribbon cables to avoid confusion and replaced the burnt out P-fet.  All that remains is to test this on the CS bot to assure we don't have another smokeout and we should be ready to put it back on Asembis.

jakunesh, 2015-03-30:
Flipping the headers did not solve the problem.  The pinout on the header is more messed up than I originally thought, the solution to this is making a new ribbon cable wire monstrosity that has the wires running to the proper place. Luckily we have just enough ribbon cable and connectors to do this

jakunesh, 2015-04-01:
As of last night the problem ADC cape was tested on Asembis with the new wires and it not only didn't catch on fire but was able to follow the line! Officially not on fire

closed 2015-04-01 by jakunesh