IEEERobotics/bot #331: List of things to bring on SEcon trip.

AhmedSamara, 2015-04-01:
Nothing is too obvious to list explicitly for check off. Don't trust our adrenaline fueled and sleep deprived brains of pre-SEcon to remember everything as we're packing up.

* Competition bot.   
* CS bot (incase of parts)   
* Electrical tape.  
* Duct tape.  
* Assorted tools.

whiteskulleton, 2015-04-07:
Dremel
Wooden box

jakunesh, 2015-04-08:
zip ties
extra wires
crimp gun
crimper
crimps
made wires
current work

SeanKetring, 2015-04-08:
CS (Ahmed's) Laptop
All cables for beagle bone
Hand held meter
Stuff for charging the batteries

jakunesh, 2015-04-09:
GPS
Soldering Iron and Solder
snacks
back up regulators
spare IR arrays
spare stepper motors and controllers

AhmedSamara, 2015-04-10:
* Phone chargers.  
* (If anyone has a car adapter please bring it)

closed 2015-04-15 by AhmedSamara