IEEERobotics/bot #332: Find better way to export and assign GPIO's.

AhmedSamara, 2015-04-01:
Currently we assign GPIO and PWM pins in the `setup_all.py` script.  
Daniel gives a good explanation of how it works in issue #291.   
This method works so far, but there's always been some confusion because of the fact that you have to assign it in two places: in the config file and in the setup file.  

Ideally, for the sake of clarity it would be something that you change once so that no one double assigns a signal.   

This is the first time this has become a problem, because we've never had so many inputs and outputs going on at once before.

This can wait until SEcon for now, so I'm closing and shelving it until then.

AhmedSamara, 2015-04-04:
Another thing the library isn't doing at all right now is changing the device tree overlay.  
We've gotten incredibly lucky up until now because we've never had to do that. 
I've created and configured a device tree overlay file on @SimonSaysServo branch, in `/bot/setup/fs-common/root/startup/device_tree_overlay.dts`

and it will have to be compiled manually with `dtc -O dtb -o pinctrl-test-0-00A0.dtbo -b 0 -@ device_tree_overlay.dts`

dfarrell07, 2015-05-08:
Ansible is a good tool to look into for this.

jasteve4, 2015-05-31:
This is being moved over to a wiki.

AhmedSamara, 2015-05-31:
Has this been resolved? Why is this closed?