IEEERobotics/bot #333: No file or directory for gpio pins

ChadZhongqiangChen, 2015-04-01:
When I started the server on the bot and called a command, the following error came up. Does it mean that I need to reassign the pins or create the file for gpio pins?

"
ctrl_server.py | call_method | 264 | WARNING | Exception: '[Errno 2] No such file or directory: '/sys/class/gpio/gpio44/value''
--> {'msg': "Exception: '[Errno 2] No such file or directory: '/sys/class/gpio/gpio44/value''", 'type': 'error'}
"

AhmedSamara, 2015-04-01:
It means the pin isn't being exported.   
They're being assigned in `config.yaml`,  but the setup script actually has to begin dealing with them.  
in `bot/setup/fs-common/root/startup/setup_all.sh` , add the lines of code for the gpio number you're trying to export.   
`load_gpio <your number here`

AhmedSamara, 2015-04-01:
Oh, and then run the setup script with `./setup_all.sh`.

Daniel provides a more detailed explanation of what this script does in #291 .

AhmedSamara, 2015-04-01:
Actually, the correct command is `./setup_bone.sh`.

`setup_all.sh` exports gpio's, but does not copy the setup scripts into the init.d directory.

closed 2015-04-01 by AhmedSamara