IEEERobotics/bot #341: Connection timeout on issuing follow commands.

AhmedSamara, 2015-04-05:
Starting the server works normally, but when I issue the follow_ignoring_turns commands, this is what I get.

```
bot$ follower follow_ignoring_turns
ctrl_server.py | call_method | 261 | WARNING | Exception: '[Errno 110] Connection timed out'
--> {'msg': "Exception: '[Errno 110] Connection timed out'", 'type': 'error'}
bot$ follower follow_ignoring_turns
ctrl_server.py | call_method | 261 | WARNING | Exception: '[Errno 110] Connection timed out'
--> {'msg': "Exception: '[Errno 110] Connection timed out'", 'type': 'error'}
bot$ 
```

This is on branch `pilot_dev`, on the latest commit.

Another strangle and possibly related error, is an apparent keyError.    
```
dmcc_motor.py | __init__ | 56 | ERROR | Bad motor definition for mo[0/417]ack_right'
Traceback (most recent call last):
  File "./bot/server/ctrl_server.py", line 317, in <module>
    server = CtrlServer(sys.argv[1])
  File "./bot/server/ctrl_server.py", line 114, in __init__
    self.systems = self.assign_subsystems()
  File "./bot/server/ctrl_server.py", line 133, in assign_subsystems
    self.follower = Follower()
  File "/root/bot2014/bot/follower/follower.py", line 35, in __init__
    self.driver = mec_driver_mod.MecDriver()
  File "/root/bot2014/bot/driver/mec_driver.py", line 35, in __init__
    self.motors = DMCCMotorSet(motor_config)
  File "/root/bot2014/bot/hardware/dmcc_motor.py", line 57, in __init__
    raise KeyError
KeyError
```

I tried going back to old and previously working commits and still got this error, and I can't find anything wrong with the config dicts.

@dfarrell07 any ideas?

dfarrell07, 2015-04-05:
One issue is that we're loosing information about the key error by not re-raising the exception correctly.

```
            try:
                self.motors[name] = DMCCMotor(
                    dmccs[conf['board_num']], conf['motor_num'], invert)
            except KeyError:
                self.logger.error(
                    "Bad motor definition for motor: '{}'".format(
                        name))
                raise KeyError
```

The `raise KeyError` part raises a new KeyError exception, instead of the old with with useful debugging info. Should be a simple `raise`.

Once you have that, you might get some useful info. Still digging...

dfarrell07, 2015-04-05:
Also, this is a bit confusing:

```
dmcc_motor.py | __init__ | 56 | ERROR | Bad motor definition for mo[0/417]ack_right'
```

Is the part `mo[0/417]ack_right'` what you're actually seeing, or was that a copy/paste error?

AhmedSamara, 2015-04-05:
Oh yeah, that's a copy paste error. I believe it said motor: back_right

dfarrell07, 2015-04-05:
It'd be helpful to log the value of `dmccs` returned by this code.

```python
        if not self.is_testing:
            dmccs = pyDMCC.autodetect()
            self.logger.debug("Found %d physical DMCC boards" % len(dmccs))
        else:
            self.logger.debug("Skipping autodetect due to test mode")
            dmccs = defaultdict(
                lambda: pyDMCC.DMCC(
                    0, verify=False, bus=None, logger=self.logger))
```

Current best guess is that the `dmccs[conf['board_num']]` part is responsible for the key error.

dfarrell07, 2015-04-05:
> Current best guess is that the dmccs[conf['board_num']] part is responsible for the key error.

And that the key error is responsible for the connection timeout. I think it's a low-level timeout, not a ZMQ-level one (which I've never seen fail the way you described).

AhmedSamara, 2015-04-06:
After fixing the KeyError raising the error is now:    
```
ack_right'
Traceback (most recent call last):
  File "./bot/server/ctrl_server.py", line 314, in <module>
    server = CtrlServer(sys.argv[1])
  File "./bot/server/ctrl_server.py", line 113, in __init__
    self.systems = self.assign_subsystems()
  File "./bot/server/ctrl_server.py", line 132, in assign_subsystems
    self.follower = Follower()
  File "/root/bot2014/bot/follower/follower.py", line 35, in __init__
    self.driver = mec_driver_mod.MecDriver()
  File "/root/bot2014/bot/driver/mec_driver.py", line 35, in __init__
    self.motors = DMCCMotorSet(motor_config)
  File "/root/bot2014/bot/hardware/dmcc_motor.py", line 52, in __init__
    dmccs[conf['board_num']], conf['motor_num'], invert)
KeyError: 0
```

AhmedSamara, 2015-04-06:
A possibly related issue, but I can't seem to run an `apt-get update` (already root, adding sudo does nothing but I've tried)     

```
Ign http://ftp.us.debian.org wheezy/non-free Translation-en_US           
Fetched 4,176 kB in 36s (115 kB/s)                                       
Reading package lists... Done
W: GPG error: http://ftp.us.debian.org wheezy Release: The following signatures were invalid: KEYEXPIRED 1587841717 KEYEXPIRED 1557241909
W: A error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://security.debian.org wheezy/updates Release: The following signatures were invalid: KEYEXPIRED 1587841717

W: A error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ftp.us.debian.org wheezy-updates Release: The following signatures were invalid: KEYEXPIRED 1587841717

W: Failed to fetch http://ftp.us.debian.org/debian/dists/wheezy-updates/Release  

W: Failed to fetch http://security.debian.org/dists/wheezy/updates/Release  

W: Some index files failed to download. They have been ignored, or old ones used instead.
```

dfarrell07, 2015-04-06:
`conf['board_num']` is resolving to 0 for `back_right` (as expected, via config), which apparently isn't a key for the dict `dmccs` mentioned above. Again, need to see the logged value of `dmccs` to know much more.

dfarrell07, 2015-04-06:
> A possibly related issue, but I can't seem to run an apt-get update (already root, adding sudo does nothing but I've tried)

This is totally unrelated. It's a product of not having run an update for too long. Just add an `--allow-unauthenticated` flag to skip the GPG check.[

AhmedSamara, 2015-04-07:
>2014-06-05 17:47:58,587 | DEBUG | dmcc_motor.py | __init__ | 37 | Found 0 physical DMCC boards
2014-06-05 17:47:58,589 | DEBUG | dmcc_motor.py | __init__ | 45 | DMCC Motor conf: {}
2014-06-05 17:47:58,592 | ERROR | dmcc_motor.py | __init__ | 59 | Bad motor definition for motor: 'back_right'

Is this what you meant?

It looks like it's not detecting any DMCC motors, and after running `i2cdetect -y -r 1`, which was incredibly slow (usually instantaneous), it's not detecting ANY i2c devices.

dfarrell07, 2015-04-07:
> Is this what you meant?

Yeah, that's perfect. As you said...

> It looks like it's not detecting any DMCC motors,

So that's the issue. May want to add a logger warning to `except` that throws that KeyError with a message like "This may indicate that the DMCC motors aren't being detected".

AhmedSamara, 2015-04-07:
Ok, I'll add that then.    

Would not being able to use the i2c bus also lead to the timeout error?

AhmedSamara, 2015-04-07:
Has this ever happened before? Losing contact with the i2c bus entirely?   
 
I have absolutely no idea how to deal with this right now. I may have somehow managed to ruin this when I messed with the device tree overlay, which I commited on a seperate branch but never tried to run on this bot.

jakunesh, 2015-04-07:
Since we are so close to competition, would simply switching out the bones solve the problem?  As far as I know the one on the CS bot is working fine and we need all the productive dev time we can get

AhmedSamara, 2015-04-07:
I don't really know what the problem is, but I think that this is worth a
shot, if it turns out that there's something wrong with the i2c bus.

dfarrell07, 2015-04-07:
> would simply switching out the bones solve the problem?

I think so.

Are you guys using I2C IR arrays? Those use to cause most of our I2C issues (something would be wrong hardware-wise, which might result in them holding the line low/high or similar). @R00ney knows more about this, might be able to help.

dfarrell07, 2015-04-07:
> I may have somehow managed to ruin this when I messed with the device tree overlay, which I commited on a seperate branch but never tried to run on this bot.

Very much doubt that this caused an issue. I don't remember hitting an I2C issue that was caused by software problems.

jakunesh, 2015-04-08:
It turns out the issue was linked to the Color sensor.  When I made the wire I didn't realize that the connector  on the PCB had a different layout than the color sensor itself. Removing it fixed the issue.

closed 2015-04-08 by jakunesh