IEEERobotics/bot #344: Rubiks cube getting caught after exit.

AhmedSamara, 2015-04-09:
[image]

The power screw can grab and turn it just fine, but after it returns the grip on the mold is so tight that it gets caught in the gripper.

AhmedSamara, 2015-04-09:
Actually, it doesn't always go in.

I think this can be solved by sanding/oiling the gripper, but I wanted to check with anyone else before I started tampering with hardware that didn't have backups.

jdekler, 2015-04-09:
I believe the best way to deal with these problems is
1. Sand or file the receiving edges of the turning mechanism into a chamfer
or radiused edge so that the Rubik's doesn't catch but instead will slide
in.
2. I believe the reason it isn't working on turning is because last night I
moved the gripper assembly over to the left (facing the bot) in order to
avoid hitting the gripper stepper motor. I moved it over too far, so the
cube isn't fully seated in the corner when it is squeezed by the grippers.
To correct this just loosen the screws holding the white arms to the frame
and move to the left far enough that the cube is secure but the mechanism
doesn't contact the left gripper motor.

AhmedSamara, 2015-04-28:
(This worked not that it ended up mattering).

closed 2015-04-28 by AhmedSamara