IEEERobotics/bot #349: Propose robotics competition as a special topics class.

AhmedSamara, 2015-04-22:
This summer, we should look at making the robotics competition a special topics class in a similar vane to the fire fighting drone challenge.  

During the meeting today, we established that most people would prefer for the robotics team to become a special topics class, which would be a good idea because we could have more than 4 people on the team getting credit for it, and it would be a less drastic change in the teams dynamic.   

The main obstacle to this is finding a professor to be involved with it, because we'd need an actual teacher as opposed to just Dr.Guptas approval and someone to be an adviser.  
To start with, we should start asking professors about what the actual process for establishing a special topics class is. A list of professors we could talk to for more information are:   

Sources for info:  
Dr.Sichitu (Successfully ran Fire fighting drones challenge for the last few semesters)      
Dr.Lenardi (She's our faculty adviser, can provide advice and/or connections to candidates).   
Senior design faculty: (they know us and have a somewhat positive opinion of us probably)     

Possible people to ask to be advisers:  
Dr.Livingston: Possible adviser pending on him agreeing. He's a new professor who taught ECE 455. I think he'd be a good candidate because despite his inexperience as a lecturer, he does seem like a good roboticist. Plus, a lot of the things that I saw he was working on for the class and the lab (like the robot arm, and a CLI for that robot arm), are things that we've worked on as well, so he could benefit from the class just as much as we could.    
Dr.Lobaton   
idk who else.

SeanKetring, 2015-04-22:
Dr. Chow, or Carlson might have some helpful insight.

AhmedSamara, 2015-04-29:
We found this website that lays out the requirements for a special topic class.  

http://oucc.ncsu.edu/course-offerings-using-special-topic-numbers   

This doesn't change too much about our plan, but some noteworthy exerts.   

>that the experimental course be proposed as a new course with a permanent number and approved through regular College and University procedures prior to its being scheduled for a third time.    

>Offering a course as a special topics using an existing special topics course number does not require approval beyond the department or other internal processes for such an offering as established by the College.

AhmedSamara, 2015-05-10:
I'm meeting with Dr.Sichitu (head of the fire fighting drones challenge) this week to discuss ideas and structure for the class. For now, here's a summary of our notes from the last meeting we had as a group.    

## Requirements for the class.  
* Write up a syllabus.  
* Grading method and class structure.   
  * Best bet is to follow ASME. They have an adviser, they regularly write progress reports
* A professor

## Learning goals.    
* Needs to be defined for any class. This will vary, but a few things will be consistent.   
  * Embedded Linux.   
  * Object Oriented Programming.  
  * Computer Vision    
  * CAD & 3d-printing.  

## Things we need to figure out.   
* How will people register for the class?  
  * Open registration? 
  * Applications like in FFDC?   
  * Requiring people to join the club for a while and then register for it the next semester.     
* Who will teach it?  
  * Hopefully we can just have an adviser like ASME to report to.   
  * We may need to get several professors (The main ECE robotics professors, Chow, Lobaton, Grant) on board to be able to make this happen.

AhmedSamara, 2015-05-13:
I wrote up a rough draft of the Syllabus. It's on the drive under 'Project proposals' directory.

AhmedSamara, 2015-05-15:
Earlier this week I spoke to Dr.Sichitu about ideas for the course, and he gave a lot of good advice, in particular the most important one being that he gave me a list of robotics professors who he said would be candidates for advisers.  (and more importantly, said that he would be willing to be the adviser if none of the others agreed to it).  

Do you guys think it would be better to hold one big meeting with all of them, or to try and meet with them individually?   
The reason I ask is because when I spoke to Dr.Livingston he expressed concern about being the only supporting professor, so it would definitely be beneficial for all of us if there was more than one professor. I don't think most professors will be willing to be on board unless we can guarantee that their time commitment will be minimal.

R00ney, 2015-05-15:
Individual meetings are probably easier, given that every professor has their own schedule to keep to.

dacox92, 2015-05-16:
When you meet with the professors individually you can explain that you are meeting with other professors and already have the backing of Dr. Sichitu for the course.

AhmedSamara, 2015-05-21:
Dr.Grant said no, Dr.Lobaton said no, Dr.Chow is unresponsive.  I think we may have to get Sichitiu.

AhmedSamara, 2016-02-24:
Dr.Gupta said that she would be willing to!

We need to re-edit our syllabus and figure out if we want to do this in the Fall or Spring