IEEERobotics/bot #350: Use mouse for localization.

AhmedSamara, 2015-04-28:
The 2016 competition depends heavily on localization, and there won't be a line to follow, so we need to find more ideas for ways to get around this.  
One method is the use of a mouse to keep track of the robots movement. We tried to use this method a few years ago but that didn't really pan out.    
This is also a good candidate for a project that can be exported as its own open-source library that will be useful for any robotics project. 

### Notes 
* We may need to invest in a decent mouse that works on any surface.  
* Most guides are on PS/2 mice, but keep in mind we have USB port and full OS tools.  

### Requirements   
* Basix X,Y co-ordinates are a basic requirement for this project.   
* velocity/acceleration information. (This is what the mouse hardware actually provides, the rest has to be integrated by the software).

### Resources   
* [Basic guide on PS/2 mouse](http://blog.jgc.org/2012/09/conversion-of-cheap-optical-mouse-to.html)       
* [Paper on odometry with multipe mice](http://botsnlinux.net/school_projects/mouse_paper_colloquium_22march11.pdf)   
* [Detailed Arduino guide](http://www.instructables.com/id/Optical-Mouse-Odometer-for-Arduino-Robot/)

Tushar1711, 2015-09-02:
Mouse will help you record your path but how will it help to trace the correct path.Do we already have the path details which we will feed the bot or  the path will be random, in which case how will mouse track the path

AhmedSamara, 2015-09-02:
The mouse won't define the path, but it will help the robot know where it
is which is necessary before it can start doing path-finding at higher
levels in code.

AhmedSamara, 2015-09-23:
@Sidd-GrizzAp3

Sidd-GrizzAp3, 2015-09-23:
I'll  be looking at this.

AhmedSamara, 2016-01-09:
@relnor  I know you were looking for something to do, would you mind looking into this?    
I forgot about it and we're running into issues with our other localization method so this would be helpful.