IEEERobotics/bot #353: Communicate between Wii remote sensors and BeagleBone black

AhmedSamara, 2015-04-28:
A wii-remote+ contains a gyroscope, accelerometer, and an IR camera (which can be used to detect walls) and costs $30.   
For now, in keeping with the philosophy of we'd just like to be able to read basic values from all these sensors and onto the BeagleBone 
## Notes   
* Wii remote plugs in over bluetooth.   
* We may use an IMU in the future, which do much the same thing, but these are much cheaper and we already have a few on hand and it would be a neat thing to show few members.   
* This has been done before, we just need to get it working on the BeagleBone.  

## Requirements   
* Read Gyroscope, accelerometer, IR camera values and send to BeagleBone.   

## Resources   
* [Several wii remote hacking videos](http://www.makeuseof.com/tag/6-crazy-interesting-or-useful-ways-to-use-a-wiimote/)   
* [Guide on interfacing with wii remote](https://www.cl.cam.ac.uk/projects/raspberrypi/tutorials/robot/wiimote/)

RyanQuinnNCSU, 2015-10-14:
Does anyone have a wii-remote, a bluetooth and a BeagleBone black  that we can non-destructively barrow?

RyanQuinnNCSU, 2015-10-14:
specifically a bluetooth dongle.