IEEERobotics/bot #354: Build competition course.

AhmedSamara, 2015-05-10:
https://docs.google.com/document/d/1ITIsL9fpTk5HKEJW1NENVrkgfCgXqONWpm0sevzeYmo/edit#heading=h.obp73v47rneu

Here are the rules including the dimensions and a 3d model of the course.    
We don't need to build the whole thing (it has two parts that are reflections of each other), and we need to be able to store it easily.

This needs to be done by the end of the summer.

SeanKetring, 2015-05-12:
Does anyone want to donate a garage or tools/supplies to this project?

SeanKetring, 2015-05-13:
From the meeting:   

Alwyn has offered his garage.  

We need a design, and some able bodies to help out.   I can help out, anyone interested in helping let me know, and we can figure out an agreeable time with everybody. 

The design has to be able to break down to be stored, as well as to the specs listed in the competition rules.  

We also need a place to store the course.

whiteskulleton, 2015-05-15:
I got all the tools necessary to build the course. Should I just go buy the materials, keep the receipts, and then get reimbursed later?

SeanKetring, 2015-05-27:
The plan is to get together in the next 2 weeks, and build it next Saturday.  We don't have a location yet

Anyone that can make it should respond to the issue, or let me or Ahmed know

SeanKetring, 2015-05-27:
We need to get alot of things that are listed on the Things to buy list, Mark if you want to hold off as we will probably be going to buy supplies sometime next week, I know @jasteve4 has volunteered the use of his truck to transport

whiteskulleton, 2015-05-27:
I haven't started on this issue yet. I can try on making the drawings with the correct dimensions and such. We need to make sure we plan out how we are going to do this with regards to interlocking the components and disassembly. My garage is more than acquitted doing the construction but it is about 35 minutes away from NCSU. Where are you guys planning on doing the construction?

SeanKetring, 2015-05-28:
If you want to come up with a plan bring it on Saturday and we can take a look, we have somewhat of a design already, but there are fuzzy details that could be worked out.  

@PaladinEng has given us free reign (kidding) of his garage for Saturday June 6th.  We are planning to buy the materials this Saturday probably a bit before 4, there is also a lowes right next to Alwyn's house, so I hear

whiteskulleton, 2015-05-28:
Can you post an address of where this is taking place?

PaladinEng, 2015-05-28:
I don't want to post my address. I'll make sure it get to all who need it.
My house is near the intersection of NC55 and I-40, not far from the
Southpoint Mall

jasteve4, 2015-05-31:
We bought a lot of the raw materials today at Home Depot. We still need to pick up the red paint form Lowes before next Saturday. 

Thanks to all that helped out today.

SeanKetring, 2015-06-03:
Everyone is doing cut plans for various parts of the course:

@jasteve4 agreed to do a cut plan for the barge section
@seanketring will do the rail car section 
@kennychuang has agreed to do the Tunnel and Rail car section 
Brett has agreed to do the floor and walls 

Cut plans will be designed from the materials we purchased at Home Depot, and will be presented to the group on Saturday at hackday

On saturday we will also paint the plywood black

Great work everybody see you on Saturday

whiteskulleton, 2015-06-03:
Are there any other tools or materials you would like me to bring? I got some great right angle clamps for setting up wood. Would we like an extra power hand saw?

SeanKetring, 2015-06-09:
Lots of good work done on this last Sunday.  

Still to do:

Dowel the course after discussing reflection methods 
Decide a method of connecting the two halves together
Paint the tunnel (yellow), barge (grey), and truck (blue)
Paint the last bin and the remaining blocks red 
Paint the rails white 

This has hack day written all over it.

whiteskulleton, 2015-06-20:
So I have been talking with Dr. Jovanovic, one of the coordinators of Southeastcon and she recently emailed me:
June 14, 2015:
"I will have to edit the playground, they want the smaller sized one. I will have to do it in next three weeks."
I asked for some clarification on what that meant and she replied:
"Do not build the course until the rules are official
They are asking for the smaller course"

What do we need to do?

SeanKetring, 2015-06-23:
We have red paint now assuming they don't change the colors we should be good with paint

PaladinEng, 2015-06-28:
we need to tell her that we've already built the course.

SeanKetring, 2015-06-28:
Sent her an email explaining our situation

SeanKetring, 2015-07-24:
Here is what Dr. Jovanovic said:

"Sean,
To answer your question.  Yes there will still be a left and right side course.  The problem that has occurred is that we have been forced to have this years rules reviewed by what feels like no less than 20 different people because of all the BS that happened last year.
In this last round of reviews there was a concern that the track was "too big", when, in fact, the actual playing field is no larger than it has been for past competitions.  The illusion that the track was too big was caused by the fact our drawings had both the left and right tracks shown on the same drawing.  To solve this issue we have updated our drawings to show separate left and right courses.
The idea was that both the left and right designs could be implemented on the same playing field if the tracks are designed with modular components.  This would make it much easier for teams to build and transport the test tracks.
I hope this clears things up.
It is important to realize that we have stated all along that the rules would not be "locked down" (taken out of draft form) until August 15th.  After August 15th all changes have to be approved by the steering committee and logged on the website.  This date was chosen because of the all the additional reviews that have been forced upon us.
Sean" 

sounds like we're in pretty good shape.

SeanKetring, 2015-07-24:
We now need to plan a hackday where we finish painting the course.   It is an end of summer goal, and one that is low hanging fruit.  

 Given that I will be out of town for a few weeks, I will not set a date just yet.

closed 2015-09-01 by SeanKetring