IEEERobotics/bot #358: Summer meeting times.

AhmedSamara, 2015-05-13:
Please fill out this survey with what meeting times work best for you.   

(And also be sure to put your name in so that we know who has responded).

http://whenisgood.net/dje98ry

AhmedSamara, 2015-05-21:
Here are the [results of the survey](http://whenisgood.net/dje98ry/results/4atknwh).

As always nothing works for everyone, but here's my suggestion:

Meeting: Tuesday @5:30, all-hands meeting @6:30.  

Hack day: Sunday @4:00. (It needs to be a weekend so that people working out of town can occasionally drop by).   

Keep in mind that hack-days are intended to be optional. If you can't make it to those specific times, I hope that you'll work on things whenever you do have time. Hack days are just meant to make it easier for people to collaborate and not be in the lab alone when they're working.

cdyoder, 2015-05-26:
Proposed summer meeting time changes:

Tuesday @ 7:00 pm for all hands meeting
Saturday @ 4:00 pm for hack day

Effective 5/26/2015

closed 2015-05-26 by cdyoder