IEEERobotics/bot #360: Make bot lighter.

AhmedSamara, 2015-05-21:
One thing that makes controlling the robot accurately so difficult is the fact that in order to get it moving at all, we have to go at a fairly high speed. (Due to torque/speed constraints).

In the 2016 competition, this is going to be even worse because without a line it'll be harder to tell things like when we've gone too far.  

We're currently using open-beam pieces to build the robot, which are great for prototyping, but I think we should consider using something lighter next year.

@whiteskulleton @psychomugs @jdekler , opinions?

dfarrell07, 2015-05-21:
You  may want to consider weighing many of the bot's components (we have two types of scales), to give you some data about what's contributing most to the problem. There might be some low-hanging fruit optimizations (like a small battery for actual runs, large one for dev time).

whiteskulleton, 2015-05-22:
Reducing the weight of the robot has to do with optimization which is usually done after we have a robot that can do the competition. I have always been designing the robot with weight considerations in mind. Furthermore, we are already using very lightweight materials, aluminum and plastic, for construction. I agree with Daniel that weighing the components will help us evaluate what needs to be changed but putting a big emphasis on reducing weight could hurt us functionally.
The real issue here is that you are having a hard time accurately controlling the robot. There are different causes for this and weight is just one of them. Looking at the past three competitions it seems that our robot has weighed over 10 lbs in each of these cases. That being said, I do not see this upcoming robot to be any different. Furthermore, we are going to be picking up wooden blocks which is just going to add more weight to the robot. Therefore, I think we need to reevaluate the hardware that we are using to move the robot and upgrade it so that it can take the weight and give you the accuracy you desire.
I think it would be a good idea to brainstorm other solutions to this issue. In industry, we have problem solving workflows that help us determine the best course of action when faced with these issues.
Here are a few ideas to get the ball rolling:
- Get more motors with a bigger gear box on them so it can supply more torque at the expense of speed.
- Another idea is to replace the motors with a different type of mechanical system like a servo or a stepper motor as those devices have other benefits and drawbacks.
- Put the wheels on bearings and use an indirect drive method to move the robot. I already have the parts to test this just need a lathe with a tailstock to drill out the part.
- Use lighter materials such as titanium or carbon fiber
- Use a different method of movement such as treads or regular wheels to drive the robot.

These are just a few ideas to fix this issue and each of these things needs to be evaluated for its feasibility.

SeanKetring, 2015-05-27:
We are looking at using an arylic base to help with this here is a link to an issue about this

https://github.com/IEEERobotics/bot2014/issues/364

cdyoder, 2015-06-23:
I've been thinking about this issue for a few weeks now and while I'm all for cutting weight, I think that is a secondary concern. Long story short, the more weight we try to cut by optimizing designs and searching for lighter chassis material, the more money and time we spend (case and point, look at how much time and money the airline industry is pumping into finding a material more resilient than titanium, more light than carbon fiber, and stronger than aluminum). 

Personally, I agree with Mark's earlier point about adequately sizing motors to suit the needs of the robot at weight and to modify the drive train to reduce the overhung load on the motor bearings to increase efficiency. I think this should be our primary mechanical focus.

That being said, if the club wants an acrylic base for prototyping or testing motor control theory or directional algorithms, I'm all for that and am willing to discuss the possibilities if anyone else is interested. 
-CY

SeanKetring, 2015-06-28:
This is included in the end of summer milestone for the mechanical team,  

next steps:
weighing components of skelebot (cs bot)  developing a more complete idea of where the weight is coming from

SeanKetring, 2015-10-02:
This bot no longer exists.

closed 2015-10-02 by SeanKetring