IEEERobotics/bot #363: Reminder: Clean out the dead branches of code and make a new repository for 2015!

cdyoder, 2015-05-26:
Please and thank you (we have new members from comp sci and comp engineering who are very interested in our code base).

AhmedSamara, 2015-05-27:
We're going to continue using the same repository, just tagged with a new version number release.

dfarrell07, 2015-05-27:
You could change the repo name without much breakage.

AhmedSamara, 2015-06-12:
I've changed the repo name to 'bot' unless anyone has a suggestion for a better name,

dfarrell07, 2015-06-13:
+1 to bot (original name)

SeanKetring, 2015-06-28:
We still need to clean the dead branches, there are so many that its tough to tell which ones are useful

kvijay1995, 2015-06-29:
I could help closing the branches I started. If Ahmed does some of his we would be in pretty good shape.

AhmedSamara, 2015-06-29:
Closed all defunct branches.

closed 2015-06-29 by AhmedSamara