IEEERobotics/bot #368: Recruiting Presentation

whiteskulleton, 2015-06-19:
I think we need to set up a presentation for recruiting in the fall. The company I am working for is doing a presentation skills workshop on Thursday, June 25 and I signed up for it. They ask me to provide a topic and I thought this would be a good opportunity to create a presentation for recruiting for our robotics group. So I was wondering what content you guys want me to include in the presentation.
 
Also if you guys could provide some images and videos of the robots over the years that would be great.
 
I found this: https://plus.google.com/photos/+DanielFarrell/albums/5998182532799064833?sort=1

SeanKetring, 2015-06-19:
Check under the 'team photos' folder on the google drive

SeanKetring, 2015-06-28:
Mark because you self assigned this task, are you going to post a draft of a recruiting presentation, or how is this issue going to get resolved?  

Do you have any helpful information from the workshop?

whiteskulleton, 2015-08-21:
I did get feedback and information on how to give an effective presentation but I couldn't make a recruitment presentation. So I do not have a recruitment presentation set up and that needs to be created.

AhmedSamara, 2015-08-29:
https://docs.google.com/presentation/d/1pk08Sy7VgA_Rcaw5AojpDEi4cutLJw88xxTuq6SWeLc/edit#slide=id.p

Here's a powerpoint if anyone wants to edit it.

I'm thinking a lineup like this:

  
* Intro to the team and members   (2-3 slides)
* Past competitions   
* slides 6   
* This years competition    
* Team resources

closed 2015-09-10 by AhmedSamara