IEEERobotics/bot #371: Getting Skelebot up to speed and under control

SeanKetring, 2015-06-28:
We need to test the new CS bot to make sure that it is able to be driven around and controlled.  Currently it has not been tested since its reconstruction.

jonathonwpowell, 2015-07-01:
got bot to move with the app, but the NICs are crashing often during operation.

SeanKetring, 2015-07-08:
Found an electrical problem the power board that were causing the NICs to crash.  Going to resolder a couple ground connections today, and see if this solves the problem

SeanKetring, 2015-07-11:
Found a faulty connection that was cutting out power, outside of dark power, in a broken fuse holder on the switch board.  Replaced this and now the connection is secure.  I believe this will solve the issue with crashing.

SeanKetring, 2015-07-24:
Skelebot is able to be driven and functional

closed 2015-07-24 by SeanKetring