IEEERobotics/bot #375: Simplifying BeagleBone setup (not just dev environemtns) using Vagrant?

AhmedSamara, 2015-07-22:
@dfarrell07 

Is there any reason to not do our configurations through a Vagrant machine, where the hardware-related files are just mounted to the VM, and just do all of our coding through that VM?

It seems to me like it would just be a really easy way to get new BeagleBones up and running quickly.

dfarrell07, 2015-08-04:
I don't understand the proposal, sorry.

dfarrell07, 2015-08-04:
I think the best way to both configure a bone and a Vagrant box for off-bone dev work is to create an Ansible role that does the proper config, then apply it to real bones or the Vagrant box (via the Vagrant Ansible provisioner) as-needed.

dfarrell07, 2015-08-04:
That'll give you the same codepath for both installs and put all your config management in a single, very easy to work with, modern/awesome/good to learn tool.

closed 2015-09-02 by AhmedSamara