IEEERobotics/bot #376: Fixing travis builds

kvijay1995, 2015-08-01:
I noticed that there exists a branch named "fixing_travis_errors" and that each travis build has three different jobs namely "docs", "pep8" and "py27". 

I also noticed that we have a different issue where we're actively trying to convert the code base to python 3 #295. If this is the case is it better to not try and fix "py27" job since the conversion will effectively replace that with a "py30" job or something along those lines @AhmedSamara @dfarrell07 ? Does the same thing apply for fixing pep8 errors, since the formatting will change when we migrate to python 3?

dfarrell07, 2015-08-04:
> If this is the case is it better to not try and fix "py27" job since the conversion will effectively replace that with a "py30" job or something along those lines @AhmedSamara @dfarrell07 ? 

I'd try to get the Python 2.7 job passing, then worry about the Python update. Once you have that stable, you can upgrade to Python 3 and fix anything else that pops up. Doing it in the reverse order sounds harder, imho.

I don't think the PEP8 job is Python version specific, I think `flake8` handles both without intervention from us.

kvijay1995, 2015-08-11:
Okay I've been working on fixing the py27 job but I can't get the tests to run on my local machine exactly the way the tests run on travis's servers. And by running the tests on my local machine, I mean these two different ways:

<b>Method 1:</b>
<ul><li>docker run -ti ieeerobotics/bot bash</li>
<li>cd /opt/bot</li>
<li>tox -epy27</li></ul>

The errors output by this method are very different from the errors I see on the travis job (job# 419.1). Also the image seems to be outdated on dockerhub which might be the cause of this not working properly.

<b>Method 2:</b>
<ul><li>(navigate to bot directory)</li>
<li>python setup.py test (OR) tox -epy27</li></ul>

The errors output by this method are different from both Method 1 and the travis build errors. (Here's the gist to the errors output by this method currently in case you're curious: https://gist.github.com/kvijay1995/43364dbfa71b0d152489)

And here's a link to the travis job for convenience: https://travis-ci.org/IEEERobotics/bot/jobs/74663713

A lazy workaround for this would be to check the travis job for the error, fix them locally and push them to github, rinse and repeat until all the errors are fixed and the travis job succeeds. But doing so will involve a lot of unnecessary commits. I've never actually run tests and fixed bugs with this python/tox/docker + tox so I'm having trouble getting the local tests to look like the tests run on the travis server and I'd like some help trying to rectify this.

NOTE: I am using the fix_travis_errors branch to fix these errors.

dfarrell07, 2015-08-11:
ACK, I'll take a look at this tomorrow. Looks like a good start, these are
the normal problems everyone hits with testing/CI. :)

kvijay1995, 2015-08-11:
Thanks Daniel! I forgot to mention something else that might be important here. The config.yaml file used to look like this:

```bash
testing : true
```

Last year we decided to change it to something like this:
```bash
test_mode: {
   color_sensor : true,
   <snip>
   etch_a_sketch: true
}
```

I believe we did this so that we could test out individual pieces of hardware connected to the bone without having to rely on stuff like the IR arrays being connected to the bone. Anyways we did not run any unit tests last year and this didn't seem to be an issue then, but all the unit tests right now seem to be referencing the "testing : true" field which is evident if you look at the errors in the travis job: https://travis-ci.org/IEEERobotics/bot/jobs/74663713

However it's not as simple as just adding this field back in, since the unit tests actually build various objects like motor, wheel gun, etc and some of these objects (like motor, DMCC) rely on the "test_mode : {...}" dict. This process is still messy and I'm working on figuring out a way to 
1) Run tests with simulated hardware (on the local machine/bone), 
2) Run tests with individual pieces of hardware on a bone and 
3) Run the fully-functional server on the bone. 

Anyways I just wanted to give you a heads up that this process is still broken. (I would appreciate any suggestions/help on this topic too since I'm learning my way around the testing process)

AhmedSamara, 2015-08-11:
Ah, a lot of that is my fault.

At the time that we vhanged how it worked I just did it as a hack and failed to go back and change everything to rely on the new structure due to more immediate priorities taking precedence.

dfarrell07, 2015-08-11:
> Method 1:

You can use `docker run -ti tox -epy27`, FYI.

I'm seeing 8 unexpected failures, all of which are the same bug. I need to revisit it or dig up some old Issues, but basically the simulated pins we use to fake hardware are files, and we need to make sure they are created in the test setup before we try to use them in the test. That should be handled by [this part](https://github.com/IEEERobotics/bot/blob/master/tests/test_bot.py#L18-L33) of the `TestBot.setUp` method (`TestBot` should be the parent class of all our tests, which makes `TestBot.setUp` run before every test).

> Method 2:

> (navigate to bot directory)
> python setup.py test (OR) tox -epy27

There are a few reasons this would fail. One, you need all the dependencies setup on your dev system (that's why we use the Vagrant box and Docker image, to package up those deps for you to share). You'd also need to fix the conditional test vs real hardware logic described correctly above.

> However it's not as simple as just adding this field back in

All you should need to do is convert everything to the new fine-grain simulation method (fixing Ahmed's hack, basically).

I suggest trying the test in the Vagrant env too, idk what the status is there but good to figure out and fix. :)

dfarrell07, 2015-08-11:
The current Travis error is an easy one, its failing at `self.orig_test_state = self.config["testing"]` because of the test simulation flag changes you described above.

> A lazy workaround for this would be to check the travis job for the error, fix them locally and push them to github, rinse and repeat until all the errors are fixed and the travis job succeeds. But doing so will involve a lot of unnecessary commits.

You could do that on a branch, then squash (via `git rebase --interactive`) your pile of commits into a few reasonable ones before merging into master.

kvijay1995, 2015-08-11:
@AhmedSamara we'll fix it now! :)

@dfarrell07 thanks for looking into this daniel. I must alert you however to the fact that the automated docker builds seemed to have stopped working. Looking at the build status [here]
(https://hub.docker.com/r/ieeerobotics/bot/builds/), the last automated build seems to have occurred at 2015-04-17T03:10:59.624123Z (The build on 2015-08-01T05:33:05.005191Z was triggered by me manually). I've also done docker pull ieeerobotics/bot and the status of the repo in that image does not reflect our latest master branch repo's status.

Do you think renaming our repo from "bot2014" to "bot" could've stopped these automated builds? anyways I will use the lazy workaround method for now to fix these testing errors.

dfarrell07, 2015-08-11:
> Do you think renaming our repo from "bot2014" to "bot" could've stopped these automated builds?

Yep, that's very likely what broke it, good call. A quick look confirms that it's tracking the `bot2014` repo. Do you have the permissions to that org required to make the changes you need, @kvijay1995?

kvijay1995, 2015-08-11:
>Do you have the permissions to that org required to make the changes?
@dfarrell07 I'm not part of the owners team on github but I am part of the owners team on dockerhub. I'm unsure of which org I'd need permissions for to make the required changes.

dfarrell07, 2015-08-11:
> [EDIT] I'm guessing I need admin rights to the repos on github to fix this issue, since I've exhausted the options on dockerhub and I can't seem to find where the dockerhub repo says it's tracking the bot2014 repo.

Yeah, it's actually pretty hidden. DockerHub's UI could use some work. Mousing over the "Source Project" [here](https://hub.docker.com/r/ieeerobotics/bot/) shows it, clicking on that link redirects (via GitHub) to the correct repo.

I'm guessing the "right" way is to just nuke the current build and start another `ieeerobotics/bot` one from the current repo. Again, DockerHub's web UI isn't obvious, so do what seems right. @AhmedSamara can grant rights everywhere, I think.

kvijay1995, 2015-08-11:
> I'm guessing the "right" way is to just nuke the current build and start another ieeerobotics/bot one from the current repo.

While trying to do this, I think I nuked the repo by mistake (my bad guys), but I've added a new repo with the same name and updated tracking details (now I hover over the "Source Project" and it shows the right repo name). I hope this fixes the builds. [Here's] (https://hub.docker.com/r/ieeerobotics/bot/builds/) the updated build details.

kvijay1995, 2015-09-18:
Me and @jonathonwpowell are on this. We are going to remove code that will not be used from last year and add a new tag. Then we will re-run the tests until everything is fixed with the possible exception of the pep8 formatting since that could be a more tedious, drawn-out task.

dfarrell07, 2015-09-18:
There are automated tools for making a good first pass at correcting
formatting to comply with PEP8.

kvijay1995, 2015-09-18:
Ok will look into that. Thanks @dfarrell07

kvijay1995, 2015-10-01:
Gentlemen, all errors have been fixed (including pep8 syntax). Let's all make sure the green passing badge stays that way :)

closed 2015-10-01 by kvijay1995

dfarrell07, 2015-10-01:
> Gentlemen, all errors have been fixed (including pep8 syntax). Let's all make sure the green passing badge stays that way :)

Wow, that's a pretty big deal! Great job!

Sidd-GrizzAp3, 2015-10-01:
HEY! I WAS SUPPOSED TO RUN THE PEP8 SCRUBBER! NOOOOOOOOOOOOOOOOO @kvijay1995