IEEERobotics/bot #378: Fall 2015 recruitment.

AhmedSamara, 2015-08-20:
Welcome back everyone.

Now that the fall semester has started, we need to start our yearly active-recruiting drive.

There are a few complications that we have to figure out this time. Mainly in regards to how we'll be doing our big meetings. We may be getting kicked out of troxler entirely this year, and while we still have access for now and Dr.Gupta has given us permission to hold interest meetings there while we prepare to leave, it's good time to start phasing out our use of it.    

I think we can use a classroom, but I'll have to talk to IEEE about getting access to reserve a classroom.

## Facebook page.  
[We're on facebook now!](https://www.facebook.com/pages/NCSU-IEEE-robotics/805210886201869).

Invite your friends, and let me know if you have any ideas for it or want admin access.

## Welcome back bash  
http://www.csc.ncsu.edu/news/1658   
We need to prepare a table for this. Let me know if any of you have any ideas for what we can do to show off. In the past we just set up the bot and let people drive it around with the app.

I think this would also be a good time to start printing off flyers for the main interest meeting.

## Main interest meeting.   
Traditionally scheduled at some point after the welcome back bash (when we've gained people for the email list).   
We should also consider putting together the presentation from #368 . I personally don't like powerpoints, but live-demos are dangerous for when they involve waiting and not-working, so having a pre-put together introduction to the team could benefit us greatly.
@whiteskulleton do you have one you want to share? If not I'll just put together a quick draft on google-drive for people to edit.

dfarrell07, 2015-08-20:
It's possible to make gifs of demos, then embed them in presentations as
reliable, lightweight demos.

There are examples of doing that type of "live" demo in this deck (the
rspec-puppet test run for example):

http://slides.com/dfarrell07/deploying-opendaylight-linuxcon

whiteskulleton, 2015-08-21:
Embedding gifs in the presentation sounds like a great idea. I second this.

SeanKetring, 2015-08-21:
Does anyone have some video of our robots for making a GIF?   You can email it to me or post it on the google drive, and I'll try to make the GIFs by Tuesday's meeting.  

@PaladinEng  I remember that you have footage of the DAGU arm/sorting robot, could we use this?   I also heard rumors of a 3d model/simulated arm, which could make an awesome video in tandem with footage of the real thing, do you have anything like this?  

As far as the welcome back booth, we need pull together a poster.  I think we should bring skele-bot to drive around,  the DAGU arm may or may not arrive in time, but I think we could have another interesting tech demo/talking point if we display the lidar map on someones laptop.  @jasteve4 can we use your lidar display?  

To the CS Guys:  is there any chance that the app might be ready for a demo next week?

SeanKetring, 2015-08-21:
Also we have fliers on our google drive that can be used for this, we just need a location and date to hold the meeting, and then I will update and print out said fliers.

marufino, 2015-08-21:
Here's the video: https://www.youtube.com/watch?v=85mXkTP5sxg
You might see a few other similar videos on my channel that are useful to you. Feel free to use them.

The rumors of the simulation are true. But it runs in tandem with the actual working arm. There is an older simulation somewhere. I can find and share the code, but someone else will have to be responsible for using it and showing it off.

I can bring in my arm as well, but I don't have time to set it up and get it working.

I can also make sandwich boards this weekend. For now, I'll do something simple that we can simply staple 8 1/2" x 11" sheets onto to announce the club meetings. If I make two sandwich boards will someone else be responsible for making the signs to put on them?
I won't make the sign boards this weekend unless someone else commits to making the signs to go on them.

SeanKetring, 2015-08-22:
> Here's the video: 5DOF Color Sorting DAGU Robot Arm

I can commit to making the signs to put on your board

PaladinEng, 2015-08-24:
I'm almost finished with the boards. I've made two. They are 15" wide by 25" tall, and painted white. I'll bring the finished board in tomorrow morning. Should I put them in HKN?