IEEERobotics/bot #383: Soldering of Dark Power and Flyback Cape PCBs

SeanKetring, 2015-09-01:
<b>Introduction</b>

We have blank PCBs that have recently arrived from Advanced Circuits, and they need to be soldered.  We have a couple for Dark Power, and 3 for the Flyback Cape.   No need to solder all of the unpopulated PCBs just one or two of Dark Power, and one of the Flyback Cape.  

This task will require learning the bare bone basics of Eagle one of the industry standard PCB design software.  

This task will also require learning some soldering skill  

<b>Notes</b>

-Components and PCBs are in the locker in a large cardboard box on the top shelf last I checked (9/1)
-Components are mixed for both PCBs
-Schematic is located on Github in the eagle-designs repository 
-eagle software is available on our shared Google Drive 
-check (and re-check)  the schematic(s) to Identify the appropriate parts 
 -when in doubt, ask, ask, ask!! 

<b/>Resources<b/>

-eagle-design repository:
https://github.com/IEEERobotics/eagle-designs

SeanKetring, 2015-09-01:
very helpful eagle tutorial:
https://learn.sparkfun.com/tutorials/using-eagle-schematic

nv4, 2015-09-24:
I can try doing this tomorrow.
Also where might I find the link to the shared Google Drive?

AhmedSamara, 2015-09-24:
@nv4 I've added you to the google drive.

SeanKetring, 2015-09-25:
Newest revision of dark power is bugged in some way, and not functional, so we will not need to solder the red pcbs in the locker.  We will be using the older revision on the board which is a purple PCB in the locker to power the bot, until critical changes are made to the newest revision.