IEEERobotics/bot #385: Kill Switch for the Motors

SeanKetring, 2015-09-01:
<b>Introduction</b>

It frequently happens that, during bot testing we loose control of the motors and they spin, uncontrolled.  This task is to create some kind of kill switch that can cut the power to all the motors without requiring we power cycle the whole bot.   This is an intermediate task, and one could require a protoboard, or preferably a PCB to be developed.  

The circuit topology and the method of implementation is up to the designer.  

Skills and understanding required:
-knowledge of circuits including switches and other components
-ability to use a breadboard 
-ability to solder a protoboard 
(stretch) 
-pcb design 

<b>Notes</b>

-components are available in Troxler, including switches
-we have our own breadboards for testing
-final deliverable should be able to be mounted to a bot

<b>Resources</b>

-How motors work
http://electronics.howstuffworks.com/motor.htm

jasteve4, 2015-09-30:
Another suggestion is to use a [rc switch](https://www.pololu.com/product/2804) connected to a rc receiver and transmitter. This would separate the code from the hardware and give us a good hardware disconnect.

jdtempl2, 2015-10-01:
I'll work on this, or at least help if someone else is already doing it

jasteve4, 2015-10-07:
Here is a [rc switch](https://www.adafruit.com/products/1095) from Adafriut.

jdtempl2, 2015-10-21:
Here's the RF receiver we should buy: http://www.adafruit.com/products/1097
And here's the fob: http://www.adafruit.com/products/1392