IEEERobotics/bot #389: 2016 Brainstorming idea dump.

AhmedSamara, 2015-09-08:
[Summarized rules powerpoint](https://docs.google.com/presentation/d/1C499PlSd5GcBbEIYxGp9EB6YHHZPqLuu7TOxXJ46thw/edit#slide=id.gc80209648_0_64)

[Official Rules](http://sites.ieee.org/southeastcon2016/student-program/)   

Feel free to dump all ideas you have either for individual subsystems or for an overall solution, no idea is too dumb or too crazy to post here. Simple ideas for what kinds of sensors to use, types of grippers, navigation and locomotion methods, or just general considerations we need to keep in mind.  Feel free to edit each post with a list of advantages and disadvantages as you come up with them but that is not initially necessary for posting here.

I would like for at least one person from each group to post here (There were 8) so I expect 8 new members to the repository to post some of their brainstorming results here.

nv4, 2015-09-08:
stereoscopic vision - two parallel cameras for depth perception, positioning (like human eyes)

jdekler, 2015-09-09:
[image]

adwiii, 2015-09-09:
Strategy/High level things:
Taking from the rules: "It is highly recommended that teams build in a hard switch to instruct their robots before each game which course they will be competing on."
Also, the containers cannot be moved during the competition.
From a strategy perspective we should complete the zones in order of B, A, C for max points.

Robot:
1:
An arm/lever to sweep the blocks from the 'barge' into a central holding area of the robot.
A conveyor belt/rollers, etc. to orient and lift the blocks. Could be a bed of corkscrews so that their spinning causes the blocks to orient lengthwise, and could be used to move blocks by spinning opposite directions.
The blocks would then slide past a camera/vision system to read the markings on the end and determine the marking.
Multidirectional drivetrain so that once in front of the containers it could simply slide between them. Or you could orient a single axis drivetrain so that the output is on the side of the bot.
2:
The robot can extend 20'', and the length of the blocks all together on the barge at each section is 19''. Have a fixed wall on one side of the blocks and have a piston, etc connected to the another wall or lever on the other side, 20'' away so that it fits around all of the blocks. Then retract the piston and bring the two walls together, sliding all of the blocks into one group. Then pull into the robot for sorting, possibly using an overhead claw with camera. This has the advantage of maintaining the orientation of the blocks for sorting, but the disadvantage of the possibly risky nature of handling so many blocks at once.

Other thoughts:
There is 1'' between the blocks. We could grab them off of the 'barge' using a simple lever pressing the block into a flat edge. This would allow us to preserve orientation, but 1'' is not much room for positioning the arms between blocks on both sides.

SeanKetring, 2015-09-09:
Grabbing Blocks:

Grabbing one at a time -vs- grabbing many at once 

Grabbing one allows for a simpler algorithm in which we grab a block and identify it, and then carry it to its location, where it is deposited.  It is also more space efficient because we only have to include space for one block at a time.  

Grabbing many will allow us to move more blocks in a shorter amount of time, but will require a more complicated storage, involving some time of categorization and memory of where blocks are being stored and what their respective deposition locations are.  

Perhaps many blocks could be grabbed and stored on the robot in a smart way that will group similar blocks together, which will require some kind of organizing scheme/mechanism and a large storage space.  Possible solution: DAGU arm grabs blocks individually and places them organized onto a retracting platform which would allow for storage of blocks to be external to the main body of the robot.

SeanKetring, 2015-10-13:
Some pictures of a brainstorm related to the overall structure of the competition bot.    This is just a running idea.  New (and senior) members are asked to contribute to the overall system design in any capacity.  

This picture bellow is a sideways facing diagram of complete bot system layout.   The running idea is that we will have two subsystems, an upper and lower bot.  The lower bot contains the full power system, navigation electronics, and the microstack.   The upper bot contains an arm for grasping 

[image]

This picutre bellow is an idea of how to design a hopper such that it would be capable of dumping blocks from the upper bot, into the loading zones.  The idea is to have a hinged edge, rotation is achieved using a servo.

[image]

This picture bellow is a downward facing diagram of the lower bot subsystems layout.

[image]

This ideas need to be refined, and modeled in solidworks or some other software.

SeanKetring, 2015-10-13:
@cdyoder please add any more information to the above post describing the design

cdyoder, 2015-10-14:
Update: 
1. Assume that "Figure 1" is the highest most figure on this string and the figures are numbered sequentially (ie, the bottom most figure as of now if Figure 4).
2. In Figure 3, the dumping hopper idea was discussed last night. It doesn't need to be a dumping hopper with a door, but rather could be a drive belt on a motor (similar to a treadmill) used for getting blocks out of the hopper. 
3. In Figure 4, the batteries will most likely not fit between the wheels. Some re-design is needed to figure out where they will go. 
4. LIDAR height needs to be played around with. If possible, can we get numbers from the LIDAR team within the next few weeks as to the optimal height of the LIDAR (basically, just have the LIDAR without the bot on the course and power it and move it manually to see what height gives us the best readings)? Those solid numbers would help a great deal with re-designing the chassis. 
5. All of this needs to be put into a CAD model so this space can be visualized easily, so if anyone would like to help out with CAD, let me know.