IEEERobotics/bot #390: Secure additional funding for the team.

AhmedSamara, 2015-09-09:
Currently, we are funded primarily through IEEE (student government and E-council appropriations), but I think it's time we set up our own funds especially since the club is expanding so much.    

New members who want to get involved, I think this is a great task to get involved in if you're interested in project management 

We also receive sponsorship from Eaglecad in the form of free licences, and from Adcanced Circuits in the form of free PCB printing, but we also need cash rewards. 
 

@jasteve4 is currently looking into setting up a wells fargo student group account, in the meantime we should also look into other sources of funding.

Current ideas:   
* Ask TI for sponsorship via free BeagleBones.     
* Ask Redhat for money. I remember last year they actually came to us to ask for a sponsorship package, and I don't believe we ever followed through with this. We need ideas for how we can carry this out. Things like "For $X we'll put redhat stickers on the robot, for $Y we'll let them name it". So we need ideas for that. @dfarrell07 do you still have those contacts?     
* Fundraisers etc in general.

SeanKetring, 2015-09-09:
One idea from yesterday, contacting sparkfun to see about getting them to supply parts at a discount

SeanKetring, 2015-09-15:
Something to ask for, getting a professional license from MATLAB for free

kvijay1995, 2015-09-18:
The Underwater robotics team said that a member from their team is in touch with Heather Daughtridge (director of external relations for ECE) who helps them network and find companies that might potentially sponsor them. It might be worth talking to her just to find out if she knows any companies that might be interested in sponsoring our team.

http://www.ncsu.edu/directory/moreinfo.php?username=hmdaught