IEEERobotics/bot #391: Teaching Classes to new members

PaladinEng, 2015-09-09:
With so many new folks interested, we discussed setting up a rotation of teaching for hack days. We also talked about getting some Arduino Uno's to assist in electronics teaching and testing.

imohame, 2015-09-09:
That's a good idea!

SeanKetring, 2015-09-09:
Possible topics for 30~60 min talks/workshops, from my ECE oriented perspective 

Using Github
Coding in python 
Eagle Cad and basic layout techniques  
introduction to microcontrollers and using them to do simple things like driving motors and servos 
Good wiring practice
Communications and protocol 
Reading a schematic 
Reading a data sheet
Designing a simple embedded system
Clocks and registers

Preferably we could combine these workshops with working on the actual bot, i.e when we need to wire the competition bot, we hold a workshop on wiring, etc.  

<b>Could any new people offer up some ideas, what is it that you want to know more about? </b>

shreyashgotee, 2015-09-17:
When do these workshops start??

TabaQ, 2015-09-17:
Coding in Python, Eagle CAD, Servos, would be great things to go over for
beginners.

RyanQuinnNCSU, 2015-09-22:
If you could show Python and how to use vagrant.

dfarrell07, 2015-09-23:
If you'd like, I can teach some CS stuff.

TabaQ, 2015-09-23:
Daniel that'll definitely be useful to all of us who are beginners in ECE
and MAE.

Sidd-GrizzAp3, 2015-09-23:
Daniel, 

I am a beginner in the Dev Team, and it would be great for you to teach us  man!  We would like to know some basic stuff.  Here is a list of topics I think would be relevant: 

1: Hierarchical software design explanation. 
2: Intro to unit testing and the library you and the legacy created. 
3: Introudction to pin-control through your library i.e. setting up a pwm, gpio, adcs, other stuff.  
4: Introduction to device-tree overlay, and how you guys used python to map the device tree to the library. 

etc. 

I was gonna learn some of this stuff, and possibly update some wikis for later use for people too. 
Others showed interest in the same thing @kvijay1995  @AhmedSamara  @jonathonwpowell

imohame, 2015-09-23:
That would be great if someone can start these classes.

dfarrell07, 2015-09-23:
> 1: Hierarchical software design explanation. 
2: Intro to unit testing and the library you and the legacy created. 
3: Introudction to pin-control through your library i.e. setting up a pwm, gpio, adcs, other stuff.

Could do these.

> 4: Introduction to device-tree overlay, and how you guys used python to map the device tree to the library.

I don't actually know a lot about the device tree overlay, couldn't teach about this.

> and possibly update some wikis for later use for people too. 

This should definitely be a desired outcome of any teaching sessions - document them so they are needed less in the future.

Maybe @SeanKetring, @AhmedSamara or someone can formalize a set of topics that would be most helpful for me to cover?

AhmedSamara, 2015-09-23:
We've just been approved for a recurring reservation of EB3 2220, every
Thursday for the rest of the semester from 6:30-8:00 for having these
classes.

PaladinEng, 2015-09-23:
I know a lot about the device tree overlay and the PRUs on the BBB. I've
done a lot of work with both over the pas year. I can teach either topic.

AhmedSamara, 2015-09-23:
@dfarrell07 
there's been some uncertainty about that, because it's hard to figure out what's best covered by a class and what's best covered by wiki's and youtube videos. I don't think it's worthwhile to recreate topics that already have excellent guides written for them, so we should focus on things specific to us. Like "Writing Python for the BeagleBone", or things that are more abstract like good project/software design, or things that are hands-on and applied, like an "Intro to using BeagleBone" class.

LaurenBrennan, 2015-09-23:
I think that having a more specific guide like what @AhmedSamara said could help a lot. Is there a wiki anywhere for anything general for those of us who need to brush up on some of our skills? Like a few links anywhere to videos/tutorials people found really helpful for this and other things in general?

SeanKetring, 2015-09-24:
Hello All,

Please fill out the following survey related to next Thursday's computer
science talk.

Please and thank you

https://docs.google.com/a/ncsu.edu/forms/d/1ALniWOdSN5gt8y9Zp8oMIIiwXSoBGFST3FAVAvhLNZ8/viewform?usp=send_form

Have a good evening,
Sean